espressif/esp-idf #199: GPIO_INTR_NEGEDGE is triggered both at Falling and Rising like if it were a ANYEDGE intr.

ricardoquesada, 2017-01-05:
Code used to setup the interrupt:

__IDF version__:  latest idf github version as of January 5, 2017
__Module used__: Sparkfun ESP32 Thing
__code__:
```c++
void setup_gpio()
{
  // setting up the input GPIO
  gpio_set_direction(GPIO_NUM_4, GPIO_MODE_INPUT);
  gpio_set_intr_type(GPIO_NUM_4, GPIO_INTR_NEGEDGE);
  gpio_set_pull_mode(GPIO_NUM_4, GPIO_PULLUP_ONLY);
  gpio_intr_enable(GPIO_NUM_4);
  gpio_install_isr_service(0);
  gpio_isr_handler_add(GPIO_NUM_4, gpio_isr_handler_up, (void*) GPIO_NUM_4);
}
```

```
// Interrupt handler
void IRAM_ATTR gpio_isr_handler_up(void* arg)
{
  BaseType_t xHigherPriorityTaskWoken = pdFALSE;
  xEventGroupSetBitsFromISR(g_pot_event_group, POT_PORT1_BIT, &xHigherPriorityTaskWoken);
  portYIELD_FROM_ISR();
}
```

```
// main loop (it is a FreeRTOS task)
void main_loop(void *arg)
{
    while(1) {
        EventBits_t uxBits = xEventGroupWaitBits(g_pot_event_group, (POT_PORT1_BIT | POT_PORT2_BIT), pdTRUE, pdFALSE, xTicksToWait);
        xEventGroupClearBits(g_pot_event_group, (POT_PORT1_BIT | POT_PORT2_BIT));

        // if not timeout, change the state
        if (uxBits != 0) {
            gpio_set_level(GPIO_NUM_21, 1);
            gpio_set_level(GPIO_NUM_5, 1);
            for (int i=0; i<1000; ++i)
                __asm__("nop");
            gpio_set_level(GPIO_NUM_21, 0);
            gpio_set_level(GPIO_NUM_5, 0);
        }
    }
}
```

scope results:
The __yellow__ signal represents the time the interrupt was generated. 
The __blue__ signal represents the signal being read by GPIO.

With `POSEDGE`, it works as expected. With `NEGEDGE`, it is triggered both _Rising_ and _Falling_.

[image]
[image]

Full test is here:
https://github.com/ricardoquesada/unijoysticle/blob/924a653f9a70ac0880541be58d8110c9ee9723a0/firmware/esp32/main/main.c

ricardoquesada, 2017-01-06:
In case it helps, the period of the signal is 500 us (microseconds).

negativekelvin, 2017-01-07:
Possible that a slow rising edge plus noise triggers a falling edge interrupt?

ricardoquesada, 2017-01-07:
yes, it could be. I tried with a different signal, a square one with the same frequency, and it worked Ok.  

The rising signal doesn't seem to have a "falling" section... but somehow it is triggering a NEGEDGE interrupt.

ricardoquesada, 2017-01-09:
More info:

The __Pink__ signal is the one that I want to evaluate. Since that type of signal was confusing the ESP32 falling signal I converted it to a digital one.

The __Blue__ signal is square signal (digital). It uses a 7404 IC to generate it. Its input is the __Yellow__ signal. The output is the square one, inverted, since it uses the 7404 inverter IC. This signal is being used as input in the ESP32.

The __Yellow__ signal represents the interrupts generated by the ESP32. The interrupts are being triggered both at raising and falling no matter what I choose: `NEGEDGE`, `POSEDGE`, `ANYEDGE`.

[image]

Very strange. The other day it was working Ok. Today no. I guess the ESP32 has some issues detecting the edges... or perhaps there is too much noise? 

Any help would be greatly appreciated. Thanks.

ricardoquesada, 2017-01-13:
it seems that it was just noise.... I guess.

I was able to solve it by converting the signal to 5v, then using a voltage divider and using the the new 3.3v signal which seems "cleaner" and now the interrupts get triggered on POS or NEG or ANY.

closed 2017-01-13 by ricardoquesada

g7wbd, 2020-07-12:
"it seems that it was just noise.... I guess.

I was able to solve it by converting the signal to 5v, then using a voltage divider and using the the new 3.3v signal which seems "cleaner" and now the interrupts get triggered on POS or NEG or ANY."

How did you convert signal ?
Thanks
David

svdrummer, 2026-02-20:
2026. Seems this is now falling edge only. What fits with the description of neg edge ie falling and pos edge = rising
https://randomnerdtutorials.com/esp-idf-esp32-gpio-interrupts/