espressif/esp-idf #200: ESP32 Windows Setup Guide - Easier version

michal037, 2017-01-06:
Article written on January 7, 2017 year.
I agree to copy this article, but you MUST to add a link to the original.
If you change something in it, write what, where and why.

I present to you the simplest method, according to me.

1. Download MSYS2 installer from: https://msys2.github.io/
I made all steps on the 32-bit version. I don't test 64-bit MSYS2, You can check at your own risk. (Possible errors)
File name: `msys2-i686-20161025.exe`

2. Install MSYS2 where You want, but be careful! NEVER USE SPACE CHAR IN THE FILES PATH.
After the installation is complete. You can run MSYS2 from setup wizard or msys32/msys2.exe program.

3. Now, you must update your MSYS2. Write these commands.
Becouse MSYS2 have got `pacman` package manager, I used it.
If the program asks for approval, type `t`. You MUST write these 3 commands!
- Update the package database and pacman:
`pacman -Sy pacman`
- Update the package database and core system packages:
`pacman -Syu`
- Update the rest.
`pacman -Su`
- Restart MSYS2

4. Write in MSYS2 console:
```
cd /
mkdir home
mkdir opt
```

5. Now you must join to `msys32/home` directory, in your windows. In my case, localisation is: `C:\msys32\home`
Now is time to download several files.
- Download `windows_install_prerequisites.sh` script from this localisation:
`https://raw.githubusercontent.com/espressif/esp-idf/master/tools/windows/windows_install_prerequisites.sh`
- Download `get-pip.py` from `https://pip.pypa.io/en/latest/installing/` page, or use this link for raw text:
`https://bootstrap.pypa.io/get-pip.py`
- The last one file is 'pyserial-3.2.1.tar.gz' or download latest version, from here:
`https://github.com/pyserial/pyserial/releases`
If you download another version, remember to change some commands where pyserial is.

Save this three files to `msys32/home` directory.
```
Michal@Michal MSYS /
$ cd /home

Michal@Michal MSYS /home
$ ls
get-pip.py  pyserial-3.2.1.tar.gz  windows_install_prerequisites.sh
```

6. Now you must install Python `version 2.7`. Write in MSYS2 console this:
`pacman -S python2` and type `t`, if program ask you about it.

VERY IMPORTANT: you must change Python name. from: `python2.7.exe` or `python2.exe` to `python.exe`.

For compatibility, write this in MSYS2 console:
```
cd /usr/bin
cp python2.exe python.exe
```

7. You need `tar` program, for unpack the file.
Write this in MSYS2 console:
`pacman -S tar` and type `t` for true.

8. Now you must run some scripts for the installation of the necessary files.
Write this commands:
```
cd /home
./get-pip.py
tar -xvf pyserial-3.2.1.tar.gz
cd pyserial-3.2.1/
python setup.py install
cd ..
./windows_install_prerequisites.sh
```
After all, You must see it:
```
************************************************
MSYS2 environment is now ready to use ESP-IDF.
Run 'source /etc/profile' to add the toolchain to
your path. Execute 'msys_shell.cmd' to launch an
MSYS terminal.

Once ESP-IDF is downloaded/checked out, set the
environment variable IDF_PATH in /etc/profile to
point to the directory.
************************************************
```
All is well.

9. Remove unnecessary files
```
cd /home
rm get-pip.py
rm pyserial-3.2.1.tar.gz
rm -rf pyserial-3.2.1/
rm windows_install_prerequisites.sh
```

10. Now, you must download `Espressif IoT Development Framework` and `Template Application`,
commands for this:
```
cd /home
git clone --recursive https://github.com/espressif/esp-idf.git
git clone https://github.com/espressif/esp-idf-template.git
```

11. Write this commands for auto set-up `xtensa-esp32-elf` and `esp-idf` path.
```
cd /home/esp-idf
chmod +x add_path.sh
echo "export IDF_PATH=\"/home/esp-idf\"" >> /etc/profile.d/esp32_toolchain.sh
echo "cd /home/esp-idf" >> /etc/profile.d/esp32_toolchain.sh
echo "./add_path.sh" >> /etc/profile.d/esp32_toolchain.sh
echo "cd /home" >> /etc/profile.d/esp32_toolchain.sh
echo "echo \"MSYS2 console for ESP32\"" >> /etc/profile.d/esp32_toolchain.sh
```

12. Restart MSYS2 and You are ready for build your first app !

---

For build your first app, follow these steps.

1. Open `esp-idf-template` directory
`cd /home/esp-idf-template`

2. Prepare project, type:
`make menuconfig`

Go to `Serial flasher config` and select for yourself `Flash SPI mode` `Flash SPI speed` `Flash size`
Go back, and select `Optimalization level (Debug)` and change it to `Release`
At the bottom menu, press `Save`, `Ok`, `Exit`.
`Exit` to return.

3. To compile application type:
`make all`

At the end you can see similar command:
```
To flash all build output, run 'make flash' or:
python /home/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/esp-idf-template/build/bootloader/bootloader.bin 0x10000 /home/esp-idf-template/build/app-template.bin 0x8000 /home/esp-idf-template/build/partitions_singleapp.bin
```

What is important? Let's see.

Now you must write you app to ESP32-SPI memory, I use `flash_download_tools_v3.4.4` from
`https://espressif.com/en/products/hardware/esp32/resources`
At the bottom in section `Tools`, you can download this program.

`--baud 115200`, you must set baud rate: `115200`
`--flash_mode dio`, set `SPI MODE` as `dio`, or the same that you selected in `menu config`
`--flash_freq 40m`, set `SPI SPEED` 40MHz

Now cherry on the cream.
```
0x1000 /home/esp-idf-template/build/bootloader/bootloader.bin 
0x10000 /home/esp-idf-template/build/app-template.bin 
0x8000 /home/esp-idf-template/build/partitions_singleapp.bin
```
You can see first hex number and then file. For better example, you must select in flash tool file named: `bootloader.bin` and set for this file `0x1000`.
This number is address in memory where app must be written.

Select Flash Size, in flash tool you see `8Mbit`, `8Mbit/8=1MB flash`, `128Mbit/8=16MB flash`.

Remember about `COM`, you can find it in `Windows Device Manager`.

Enter to `Download Mode`, and click `Start` button to flash app to memory.

I think, it's all. If something is wrong, tell me it!
Good Luck!

Kozakiewicz Michał, michal037

projectgus, 2017-01-08:
Hi Michal,

Thanks for contributing this set of steps. We're very keen to make the Windows instructions as easy as possible, however it's not precisely clear to me in what way these steps are easier than the [official getting started guide](http://esp-idf.readthedocs.io/en/latest/windows-setup.html).

As a starting point, could you please explain to me the problems you've found with the existing Windows setup guide. I'd like to understand your motivations for writing this document. Then perhaps we can work through them together to find improvements.

michal037, 2017-01-09:
I decided to deal with errors in the order. 
After removing the first mistakes, move on to the next, or it may be a situation where it will not be them.
In the latest msys2 `msys2-i686-20161025.exe` for day January 9, 2017.

Before checking the file, I please you, read the rest.
Log file: `https://raw.githubusercontent.com/michal037/esp32-for-win-logs/master/msys2_esp32.log`

---

The first error occurs here:
```
(68/75) installing mingw-w64-i686-termcap                                                              [###########################################################] 100%
(69/75) installing mingw-w64-i686-readline                                                             [###########################################################] 100%
(70/75) installing mingw-w64-i686-tcl                                                                  [###########################################################] 100%
(71/75) installing mingw-w64-i686-tk                                                                   [###########################################################] 100%
(72/75) installing mingw-w64-i686-python2                                                              [###########################################################] 100%
(73/75) installing mingw-w64-i686-python2-setuptools                                                   [###########################################################] 100%
(74/75) installing mingw-w64-i686-python2-pip                                                          [###########################################################] 100%
/
(75/75) installing unzip                                                                               [###########################################################] 100%
./windows_install_prerequisites.sh: line 28: python: command not found

Michal@Michal MSYS /
$
```

After running the script and downloaded his files. In directory `/usr/bin` is missing `python` interpreter.
Therefore the command can not be performed. From line `28`.
`python -m pip install --upgrade pip`

---

Next problem is missing directory `/opt`, script download `xtensa-esp32-elf` compilator but does not store its files.

---

The third problem is that command: `cd ~`. `35` line.
This command instead select the appropriate directory in msys2. 
Selects completely wrong from the C drive.

---

In my script, I create before it, directory `/home` and `/opt`.
`/opt` for xtensa compilator, `/home` for esp-idf and other files.

I think that this console msys2 should be only for esp32.
Becouse, there is:

from <41 to 43> line:
```
cat > /etc/profile.d/esp32_toolchain.sh << EOF
export PATH="$PATH:/opt/xtensa-esp32-elf/bin"
EOF
```

I think is good to download `esp-idf` to `/home/esp-idf` directory, Because why not?
I know that, esp-idf is constantly updated in github, I prepared for that update script.

```
# Create script for updating esp-idf
echo -en "#!" > update_espidf.sh
echo -e "/bin/bash" >> update_espidf.sh
echo "cd /home" >> update_espidf.sh
echo "rm -rf esp-idf" >> update_espidf.sh
echo "git clone --recursive https://github.com/espressif/esp-idf.git" >> update_espidf.sh
echo "echo \"ESP-IDF Update Done\"" >> update_espidf.sh
chmod +x update_espidf.sh
```

For manually updating, when user want it.

To further simplify the process, I recommend you to use this example of command. 
It is very easier and better than manually downloading the file `windows_install_prerequisites.sh`.

`curl -sS https://raw.file-xxxxxx.sh > /raw.file-xxxxxx.sh && cd / && chmod +x raw.file-xxxxxx.sh && ./raw.file-xxxxxx.sh`

Only one command to create a complete environment.

occamw, 2017-02-13:
Up to Section Two - Compiling And Flashing Applications step2. all seems to be OK.
When running step 2. "make menuconfig" following error shows up:
zconf.hash.c:166:1: error: conflicting types for ‘kconf_id_lookup’
 kconf_id_lookup (register const char *str, register size_t len)
 ^~~~~~~~~~~~~~~
zconf.hash.c:33:31: note: previous declaration of ‘kconf_id_lookup’ was here
 static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);

Please help!

occamw, 2017-02-13:
The suggestion to update gperf.git is no help!

michal037, 2017-02-13:
@occamw Try to follow if you have a well written path to the compiler. It is also possible that a bug in the sdk.

projectgus, 2017-02-13:
@occamw in future, please open a new issue for an unrelated error instead of commenting on an existing issue.

> zconf.hash.c:33:31: note: previous declaration of ‘kconf_id_lookup’ was here
> static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
>
> Please help!

The new gperf version 3.1 changed the type signature for ID lookup functions. There is a fix for this coming soon in esp-idf, but in the meantime you can entirely delete this line in `zconf.gperf` (note this is in a different file to zconf.hash.c, which is autogenerated by gconf from this source file):
https://github.com/espressif/esp-idf/blob/master/tools/kconfig/zconf.gperf#L12
... and it should build correctly.

projectgus, 2017-02-16:
> zconf.hash.c:33:31: note: previous declaration of ‘kconf_id_lookup’ was here
This is fixed in the master branch as of 578e938.

wfbrown53, 2017-02-25:
Worked like a charm. Only issue I saw was having to run the build multiple times to get all the projects built. Seems to all have been built after all that.
Thanks for the efforts Michael.

closed 2017-03-28 by igrr

projectgus, 2017-03-28:
Hi @michal037 ,

I think the fixes in 5faee6a should take care of the problems you were describing here. If you think there is anything left which doesn't work smoothly with the Windows setup, please open a new issue and let us know about it.

Angus

brownx, 2017-04-05:
The Windows setup from http://esp-idf.readthedocs.io/en/latest/windows-setup.html still not complete - at least not the pre-prepared environment. The make for the template project searches for python:
_...
CC bootloader_start.o
AR libmain.a
LD bootloader.elf
/bin/sh: python: command not found
make[1]: *** [/d/Sources/esp-idf/components/esptool_py/Makefile.projbuild:49: /d/Sources/esp-idf-template/build/bootloader/bootloader.bin] Error 127
make: *** [/d/Sources/esp-idf/components/bootloader/Makefile.projbuild:31: /d/Sources/esp-idf-template/build/bootloader/bootloader.bin] Error 2_

Tried to add the C:\msys32\mingw32\bin to %PATH% in the environment variables since there I have seen python but solved nothing.

projectgus, 2017-04-05:
@brownx there's a mistake in one of the steps, you need to run `...\msys32\mingw32.exe` not `...\msys32\msys_shell.cmd`. It was only changed in one place not both place.

Ref: https://github.com/espressif/esp-idf/issues/493#issuecomment-291690881

Fix coming shortly.

Gfast2, 2017-05-24:
@projectgus And it's not ```msys2.exe``` neither. (ps: spend about 2h to figure this out)

projectgus, 2017-05-25:
Hi @Gfast2,

I'm very sorry to hear that. From what I can see it looks like msys2.exe was incorrectly mentioned in the first post in this issue.

Rather than using the user-submitted guide found above, please use our official documentation here:

http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html

- it's been improved significantly since ths issue was first opened.

If find any shortcomings in that documentation, please open issues to report them - we'll gladly fix them.

Gfast2, 2017-05-25:
@projectgus All right. Thank you all again for the effort behind the scene! I really appriciate! 
_ps_: 
Since I'd building some industrial level device based on ESP32 right now. Is there some list to showing sold production Product based on this chip?

memory125, 2018-12-06:
It's seemingly we should open the mingw32.exe not  msys2.exe  while compiling and flashing on Windows.

michal037, 2018-12-06:
@memory125 It was my mistake. I did not notice it then. That's why I tried to figure out why it does not work.

memory125, 2018-12-06:
> @memory125 It was my mistake. I did not notice it then. That's why I tried to figure out why it does not work.

No worries, it works fine now.

diegosantos2, 2019-04-10:
Hi michal037, and thk thk thk thk thk a lot !!!
If i not get your post i never will do my project start !!!
Ir is very cool man, keep going !!!
thk again :)

hussaa1, 2020-02-05:
Hello Michal037,

I really liked your solution for setting up my ESP-IDF after failing to install properly with expressif website tutorials. Although your details are very well explained and documented with clear understanding, I am still not able to compile an app in my environment. The make all process just sits there all day long without compiling. In the end I have to just terminate the process.

Initially I tried to install everything manually and didn't succeed and then I tried to install with the long one off command set that you have provided. Before I tried the long command set, I reinstalled my MSYS2 32bit program in a different location on my Windows 7 32bit machine C:/msys2 directory without any space chars etc. 

I am getting this error and don't know what to do. Any suggestions?

$ curl -sS https://raw.githubusercontent.com/michal037/VariousDestiny/master/esp32/esp32-autotoolchain-msys2.sh > /esp32-autotoolchain-msys2.sh && cd / && chmod +x esp32-autotoolchain-msys2.sh && ./esp32-autotoolchain-msys2.sh
./esp32-autotoolchain-msys2.sh: line 1: 404:: command not found

Many thanks

michal037, 2020-02-05:
Hello @hussaa1

I wrote this issue because I wasn't aware that MSYS2 should be run via the mingw32.exe file and not msys2.exe.

Since closing this issue, the official documentation is so good that it worked for me and for most people. Script was no longer current, other files may were changed, which my guide was dependent on. That's why I deleted this script and you see this error message.

All the steps I described were in the script and a bit of not required configuration.

Maybe look for a mailing list (if it exists)? To get advice on how to fix it right now.

Now it's hard for me to help you because I haven't compiled anything for a long time for ESP32.

projectgus, 2020-02-05:
Hi @hussaa1  & @michal037 ,

> Maybe look for a mailing list (if it exists)? To get advice on how to fix it right now.

We don't have a mailing list but there is an active forum: https://esp32.com/

@hussaa1, if following the [Get Started Guide](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/) for the version of ESP-IDF that you're using is not working then please post there with some details about what you've done and what error(s) you see, and someone should be able to help.

Note that in ESP-IDF V4.0 and newer, we no longer require or recommend MSYS2/MINGW32 for Windows users - the Get Started steps will guide you through a different installation method.

commit f5aee6a6 (2017-03-28): windows config: Update Windows setup script, pre-installed image, instructions.

Add notes about MSYS2 package mirrors & HTTP proxies for users in China.

Closes #260 https://github.com/espressif/esp-idf/issues/260
Closes #281 https://github.com/espressif/esp-idf/issues/281
Closes #200 https://github.com/espressif/esp-idf/issues/200
Closes #430 https://github.com/espressif/esp-idf/pull/430
Accomplishes similar goals to #264 https://github.com/espressif/esp-idf/pull/264