espressif/esp-idf #201: Meditation Error of type InstrFetchProhibited occurred on core  0

Gianpy86, 2017-01-07:
Hi Experts,

I'm trying to run my first esp-idf project, here my code:
```c
/* Hello World Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

void WiFiWPAEntConnect(void *pvParameter)
{
    printf("Hello WiFi!\n");
    tcpip_adapter_init();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "NoobsLand",
            .password = "",
        },
    };
    ESP_LOGI("WPA2", "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_LOGI("WPA2", "Wifi mode set");
    esp_wifi_sta_wpa2_ent_set_username((unsigned char *)"ESP32", 4);
    ESP_LOGI("WPA2", "Wifi username set");
    esp_wifi_sta_wpa2_ent_set_password((unsigned char *)"ESP32", 4);
    ESP_LOGI("WPA2", "Wifi password set");
    ESP_ERROR_CHECK( esp_wifi_start() );
    ESP_LOGI("WPA2", "Wifi start");

}

void app_main()
{
    nvs_flash_init();
    xTaskCreate(&WiFiWPAEntConnect, "WiFiWPAEntConnect", 2048, NULL, 5, NULL);
}
```

When flashing my ESP32 it goes in loop because of this error:

```

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:2364
load:0x40078000,len:3724
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (1257) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1258) heap_alloc_caps: At 3FFBD480 len 00022B80 (138 KiB): DRAM
I (1289) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1353) heap_alloc_caps: At 4009B844 len 000047BC (17 KiB): IRAM
I (1415) cpu_start: Pro cpu up.
I (1450) cpu_start: Single core mode
I (1490) cpu_start: Pro cpu start user code
D (1645) intr_alloc: Connected src 56 to int 2 (cpu 0)
D (1649) intr_alloc: Connected src 16 to int 3 (cpu 0)
D (1652) nvs: nvs_flash_init_custom start=9 count=6
D (1724) phy_init: loading PHY init data from application binary
D (1746) nvs: nvs_open phy 0
D (1774) nvs: nvs_get cal_version 4
D (1807) nvs: nvs_get_str_or_blob cal_mac
D (1845) nvs: nvs_get_str_or_blob cal_data
D (1891) nvs: nvs_close 1
I (2059) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
D (2060) nvs: nvs_open phy 1
D (2060) nvs: nvs_set cal_version 4 258
D (2182) nvs: nvs_set_blob cal_mac 6
D (2186) nvs: nvs_set_blob cal_data 1904
D (3187) nvs: nvs_close 2
I (3187) cpu_start: Starting scheduler on PRO CPU.
Hello WiFi!
tcpip_task_hdlxxx : 3ffc191c, prio:18,stack:2048
D (3212) nvs: nvs_open misc 1
D (3215) nvs: nvs_get_str_or_blob log
I (3219) wifi: frc2_timer_task_hdl:3ffc2394, prio:22, stack:2048
D (3224) nvs: nvs_open nvs.net80211 1
D (3228) nvs: nvs_get opmode 1
D (3231) nvs: nvs_get country 1
D (3234) nvs: nvs_get_str_or_blob sta.ssid
D (3239) nvs: nvs_get_str_or_blob sta.mac
D (3243) nvs: nvs_get sta.authmode 1
D (3246) nvs: nvs_get_str_or_blob sta.pswd
D (3251) nvs: nvs_get_str_or_blob sta.pmk
D (3255) nvs: nvs_get sta.chan 1
D (3258) nvs: nvs_get auto.conn 1
D (3261) nvs: nvs_get bssid.set 1
D (3265) nvs: nvs_get_str_or_blob sta.bssid
D (3269) nvs: nvs_get sta.phym 1
D (3272) nvs: nvs_set sta.phym 1 4
D (3276) nvs: nvs_get sta.phybw 1
D (3279) nvs: nvs_get_str_or_blob sta.apsw
D (3283) nvs: nvs_get_str_or_blob sta.apinfo
D (3288) nvs: nvs_get_str_or_blob ap.ssid
D (3292) nvs: nvs_get_str_or_blob ap.mac
D (3296) nvs: nvs_get_str_or_blob ap.passwd
D (3300) nvs: nvs_get_str_or_blob ap.pmk
D (3304) nvs: nvs_get ap.chan 1
D (3307) nvs: nvs_get ap.authmode 1
D (3311) nvs: nvs_get ap.hidden 1
D (3314) nvs: nvs_get ap.max.conn 1
D (3318) nvs: nvs_get bcn.interval 2
D (3321) nvs: nvs_get ap.phym 1
D (3324) nvs: nvs_set ap.phym 1 4
D (3328) nvs: nvs_get ap.phybw 1
D (3331) nvs: nvs_get ap.sndchan 1
D (3335) nvs: nvs_set_blob sta.mac 6
D (3339) nvs: nvs_set_blob ap.mac 6
I (3343) wifi: pp_task_hdl : 3ffc4ff8, prio:23, stack:8192
I (3347) WPA2: Setting WiFi configuration SSID NoobsLand...
I (3353) WPA2: Wifi mode set
I (3357) WPA2: Wifi username set
I (3361) WPA2: Wifi password set
I (3365) WPA2: Wifi start
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400ee970  PS      : 0x00060f30  A0      : 0x00000000  A1      : 0x3ffc0c10  
A2      : 0x3f405c3c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800ee970  A9      : 0x3ffc0bc0  
A10     : 0x400569b4  A11     : 0x3ffb99bc  A12     : 0x3f405d2c  A13     : 0x0000001f  
A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x00000019  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400ee970:0x3ffc0c10 0x00000000:0x3ffc0ca0

Rebooting...
ets Jun  8 2016 00:22:57

```

I'm using MacBookPro Retina 15 and ATOM just as text editor.
The flashing result:

```
Giampieros-MBP:WiFiTest gianpy86$ make flash
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x10000)...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Auto-detected Flash size: 4MB
Wrote 16384 bytes at 0x00001000 in 1.4 seconds (91.2 kbit/s)...
Hash of data verified.
Wrote 425984 bytes at 0x00010000 in 37.7 seconds (90.4 kbit/s)...
Hash of data verified.
Wrote 16384 bytes at 0x00008000 in 1.4 seconds (91.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
Giampieros-MBP:WiFiTest gianpy86$ make flash
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x10000)...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Auto-detected Flash size: 4MB
Wrote 16384 bytes at 0x00001000 in 1.4 seconds (91.3 kbit/s)...
Hash of data verified.
Wrote 425984 bytes at 0x00010000 in 37.7 seconds (90.4 kbit/s)...
Hash of data verified.
Wrote 16384 bytes at 0x00008000 in 1.4 seconds (91.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

```

Can someone help me?

igrr, 2017-01-07:
In FreeRTOS, the task function is not allowed to return, like your `WiFiWPAEntConnect` function does. If you want to end the task, you need to call `vTaskDelete(NULL);`. Alternatively, just call `WiFiWPAEntConnect` function directly from `app_main`, without creating a task — in this case you don't need `vTaskDelete`.

Gianpy86, 2017-01-07:
Ok, thankyou, something changed. 
here new code:
```c
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

void WiFiWPAEntConnect()
{
    printf("Hello WiFi!\n");
    tcpip_adapter_init();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "NoobsLand",
            .password = "",
        },
    };
    ESP_LOGI("WPA2", "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_LOGI("WPA2", "Wifi mode set");
    esp_wifi_sta_wpa2_ent_set_username((unsigned char *)"ESP32", 4);
    ESP_LOGI("WPA2", "Wifi username set");
    esp_wifi_sta_wpa2_ent_set_password((unsigned char *)"ESP32", 4);
    ESP_LOGI("WPA2", "Wifi password set");
    ESP_ERROR_CHECK( esp_wifi_start() );
    ESP_LOGI("WPA2", "Wifi start");

}

void app_main()
{
    nvs_flash_init();
    WiFiWPAEntConnect();
}
```

Now, after some reboots with this error:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:2364
load:0x40078000,len:3724
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (1257) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1258) heap_alloc_caps: At 3FFBD490 len 00022B70 (138 KiB): DRAM
I (1289) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1353) heap_alloc_caps: At 4009B844 len 000047BC (17 KiB): IRAM
I (1415) cpu_start: Pro cpu up.
I (1450) cpu_start: Single core mode
I (1490) cpu_start: Pro cpu start user code
D (1645) intr_alloc: Connected src 56 to int 2 (cpu 0)
D (1649) intr_alloc: Connected src 16 to int 3 (cpu 0)
[... 94 lines cut ...]
D (3309) nvs: nvs_set_blob ap.mac 6
I (3313) wifi: pp_task_hdl : 3ffc4688, prio:23, stack:8192
I (3317) WPA2: Setting WiFi configuration SSID NoobsLand...
I (3323) WPA2: Wifi mode set
I (3327) WPA2: Wifi username set
I (3331) WPA2: Wifi password set
I (3335) wifi: mode : sta (24:0a:c4:03:fe:30)
/Users/gianpy86/esp/esp-idf/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40082737  PS      : 0x00060c33  A0      : 0x800eeb53  A1      : 0x3ffc44b0  
A2      : 0x00000000  A3      : 0x00000048  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000048  A7      : 0x00000010  A8      : 0x80082734  A9      : 0x3ffc4490  
A10     : 0x00060c20  A11     : 0x00000062  A12     : 0x00000010  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xffffffee  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40082737:0x3ffc44b0 0x400eeb53:0x3ffc44f0 0x400d1e56:0x3ffc4510 0x400d6451:0x3ffc4530 0x400d74b0:0x3ffc4580 0x400d7b96:0x3ffc45b0 0x4009779d:0x3ffc45d0

Rebooting...
```

About the 1st error i read that some times it happen because of low power energy from USB but now i'm currently using external power source with max 2A current.

The 2nd error looks like my code to connecting to a WPA2 enterprise missed something, could be the certificate even if i don't use it?

igrr, 2017-01-07:
Regarding the second error, please check one of the wifi examples — you are missing a call to `esp_event_loop_init` ([link](https://github.com/espressif/esp-idf/blob/master/examples/03_http_request/main/http_request_main.c#L79)).

Regarding the first error, indeed, it happens when RF calibration starts, which indicates insufficient power to the module. Be sure to add sufficient decoupling capacitors (a combination of small ones and large ones) close to the module VCC pin, if you are using a standalone module.

Gianpy86, 2017-01-07:
Thank you so much Ivan for the time you using to help me. 👍  
I did what you suggest and now it start every time correctly.
I made some fix to my code that looks not ok. Here the last code that i ran:
```c
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

static EventGroupHandle_t wifi_event_group;

const int CONNECTED_BIT = BIT0;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

void WiFiWPAEntConnect()
{
    printf("Hello WiFi!\n");
    tcpip_adapter_init();

    char username[]= "ESP32";
    char password[]= "ESP32";

    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "NoobsLand",
            .password = "",
        },
    };
    ESP_LOGI("WPA2", "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_LOGI("WPA2", "Wifi mode set");

    ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_enable() );
    ESP_LOGI("WPA2", "Wifi wpa2_ent_enable");
    //sanity clean//
    esp_wifi_sta_wpa2_ent_clear_username();
    esp_wifi_sta_wpa2_ent_clear_password();
    esp_wifi_sta_wpa2_ent_clear_ca_cert();

    ESP_LOGI("WPA2", "Wifi username:%s(%i) - password:%s(%i)", username, strlen(username), password, strlen(password));
[... 100 lines cut ...]
I (3463) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (3468) WPA2: Wifi wpa2_ent_enable
I (3472) WPA2: Wifi username:ESP32(5) - password:ESP32(5)
I (3478) WPA2: Wifi username set
I (3482) WPA2: Wifi password set
I (3486) WPA2: Wifi set config
I (3490) wifi: mode : sta (24:0a:c4:03:fe:30)
D (3494) event: SYSTEM_EVENT_STA_START
I (3498) WPA2: Wifi start
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d17ab  PS      : 0x00060030  A0      : 0x800d18f6  A1      : 0x3ffbfcd0  
A2      : 0x400d17a8  A3      : 0x3ffbe160  A4      : 0x800569e0  A5      : 0x3ffbf470  
A6      : 0x00000025  A7      : 0x3ffbd740  A8      : 0x80083420  A9      : 0x3ffbfcc0  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060020  A13     : 0x3ffbf450  
A14     : 0x3ffbd798  A15     : 0x3ffbd740  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d17ab:0x3ffbfcd0 0x400d18f6:0x3ffbfcf0 0x4008378c:0x3ffbfd10

Rebooting...
```

what i forgot now? :(

Gianpy86, 2017-01-08:
Here again...

i get some step next... 

[https://github.com/Gianpy86/espressif/tree/master/ESP32/WiFi_Enterprise](url)

At this link my repo with the complete source and below the result of start:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:2364
load:0x40078000,len:3724
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (754) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (755) heap_alloc_caps: At 3FFBE560 len 00021AA0 (134 KiB): DRAM
I (759) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (770) heap_alloc_caps: At 4009C124 len 00003EDC (15 KiB): IRAM
I (780) cpu_start: Pro cpu up.
I (785) cpu_start: Starting app cpu, entry point is 0x40080bc8
I (0) cpu_start: App cpu up.
I (801) cpu_start: Pro cpu start user code
D (831) intr_alloc: Connected src 56 to int 2 (cpu 0)
D (835) intr_alloc: Connected src 16 to int 3 (cpu 0)
D (836) intr_alloc: Connected src 24 to int 12 (cpu 0)
D (879) nvs: nvs_flash_init_custom start=9 count=6
D (971) phy_init: loading PHY init data from application binary
D (977) nvs: nvs_open phy 0
D (1003) nvs: nvs_get cal_version 4
D (1037) nvs: nvs_get_str_or_blob cal_mac
D (1075) nvs: nvs_get_str_or_blob cal_data
D (1126) nvs: nvs_close 1
I (1247) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
D (1248) nvs: nvs_open phy 1
D (1249) nvs: nvs_set cal_version 4 258
D (1384) nvs: nvs_set_blob cal_mac 6
D (1389) nvs: nvs_set_blob cal_data 1904
D (2428) nvs: nvs_close 2
I (2429) cpu_start: Starting scheduler on PRO CPU.
D (1638) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (1639) cpu_start: Starting scheduler on APP CPU.
tcpip_task_hdlxxx : 3ffc2e94, prio:18,stack:2048
D (2138) nvs: nvs_open misc 1
D (2138) nvs: nvs_get_str_or_blob log
I (2139) wifi: frc2_timer_task_hdl:3ffc4970, prio:22, stack:2048
D (2144) nvs: nvs_open nvs.net80211 1
D (2147) nvs: nvs_get opmode 1
D (2151) nvs: nvs_get country 1
D (2154) nvs: nvs_get_str_or_blob sta.ssid
D (2159) nvs: nvs_get_str_or_blob sta.mac
D (2162) nvs: nvs_get sta.authmode 1
D (2166) nvs: nvs_get_str_or_blob sta.pswd
D (2170) nvs: nvs_get_str_or_blob sta.pmk
D (2175) nvs: nvs_get sta.chan 1
D (2177) nvs: nvs_get auto.conn 1
D (2181) nvs: nvs_get bssid.set 1
D (2185) nvs: nvs_get_str_or_blob sta.bssid
D (2188) nvs: nvs_get sta.phym 1
D (2192) nvs: nvs_set sta.phym 1 4
D (2195) nvs: nvs_get sta.phybw 1
D (2198) nvs: nvs_get_str_or_blob sta.apsw
D (2203) nvs: nvs_get_str_or_blob sta.apinfo
D (2210) nvs: nvs_get_str_or_blob ap.ssid
D (2211) nvs: nvs_get_str_or_blob ap.mac
D (2215) nvs: nvs_get_str_or_blob ap.passwd
D (2220) nvs: nvs_get_str_or_blob ap.pmk
D (2223) nvs: nvs_get ap.chan 1
D (2226) nvs: nvs_get ap.authmode 1
D (2230) nvs: nvs_get ap.hidden 1
D (2233) nvs: nvs_get ap.max.conn 1
D (2238) nvs: nvs_get bcn.interval 2
D (2241) nvs: nvs_get ap.phym 1
D (2244) nvs: nvs_set ap.phym 1 4
D (2248) nvs: nvs_get ap.phybw 1
D (2251) nvs: nvs_get ap.sndchan 1
D (2254) nvs: nvs_set_blob sta.mac 6
D (2259) nvs: nvs_set_blob ap.mac 6
I (2262) wifi: pp_task_hdl : 3ffc75d4, prio:23, stack:8192
I (2267) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (2272) wifi: mode : sta (24:0a:c4:03:fe:30)
D (2275) event: SYSTEM_EVENT_STA_START
I (2279) EH: 2
I (2402) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (3049) wifi: state: init -> auth (b0)
I (4049) wifi: state: auth -> init (2)
D (4050) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:NoobsLand, ssid_len:9, bssid:84:26:15:42:29:9c, reason:2
I (4052) EH: 5
I (4174) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (4175) wifi: state: init -> auth (b0)
I (5175) wifi: state: auth -> init (2)
D (5175) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:NoobsLand, ssid_len:9, bssid:84:26:15:42:29:9c, reason:2
I (5177) EH: 5
```

joostd, 2017-01-08:
I am not sure how to read these debug messages, but here's what I get on my WPA2-Enterprise network (using EAP-TTLS):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:2364
load:0x40078000,len:3724
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (788) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (788) heap_alloc_caps: At 3FFBD63C len 000229C4 (138 KiB): DRAM
I (791) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (797) heap_alloc_caps: At 4009BBD8 len 00004428 (17 KiB): IRAM
I (804) cpu_start: Pro cpu up.
I (808) cpu_start: Single core mode
I (812) cpu_start: Pro cpu start user code
I (1001) phy: phy_version: 258, Nov 29 2016, 15:51:07, 1, 0
I (2047) cpu_start: Starting scheduler on PRO CPU.
tcpip_task_hdlxxx : 3ffc113c, prio:18,stack:2048
I (2059) wifi: frc2_timer_task_hdl:3ffc2c30, prio:22, stack:2048
I (2067) wifi: pp_task_hdl : 3ffc5894, prio:23, stack:8192
I (2068) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (2073) wifi: rx_ba=1 tx_ba=1

I (2076) wifi: mode : sta (24:0a:c4:00:97:90)
I (2953) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (3599) wifi: state: init -> auth (b0)
I (3603) wifi: state: auth -> assoc (0)
I (3618) wifi: state: assoc -> run (10)
I (3618) wpa: wpa2_task prio:24, stack:6144

I (3734) wpa: >>>>>wpa2 FINISH

I (3745) wpa: wpa2 task delete

I (3746) wifi: connected with homeroam, channel 6
I (6702) event: ip: 192.168.0.225, mask: 255.255.255.0, gw: 192.168.0.1
I (13618) wifi: pm start, type:0
```

I am using the following code:

```
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "os.h"

#define USERNAME "anonymous@espressif.com"
#define PASSWORD "test11"

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    return ESP_OK;
}

void app_main(void)
{
    nvs_flash_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    wifi_config_t sta_config = {
        .sta = {
            .ssid = "homeroam",
            .password = "",
            .bssid_set = false
        }
    };
    ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_enable() );
    esp_wifi_sta_wpa2_ent_clear_username();
    esp_wifi_sta_wpa2_ent_clear_password();
    esp_wifi_sta_wpa2_ent_clear_ca_cert();
    // TODO: set CA certificate
    esp_wifi_sta_wpa2_ent_set_username( (unsigned char *) USERNAME, os_strlen(USERNAME));
    esp_wifi_sta_wpa2_ent_set_password( (unsigned char *) PASSWORD, os_strlen(PASSWORD));
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &sta_config) ); // ESP_IF_WIFI_STA ?
    ESP_ERROR_CHECK( esp_wifi_start() );
    ESP_ERROR_CHECK( esp_wifi_connect() );
}
```

Having access to the server logs usually is a great help in debugging WPA2-Enterprise issues. Also bear in mind that `libwpa2.a` fixes the outer identity to `anonymous@espressif.com`, which may or may not be a problem for your network. You may need to use a binary editor to edit that library in order to set the correct identity.

A future version of the sdk will probably include an API for setting the outer identity.

Gianpy86, 2017-01-09:
Last revision of my code below.

I checked that my enterprise is an EAP-PEAP (MSCHAPv2) and still have the issue.

As u showed now im trying with user anonymous@espressif.com. I read into the documentation that for MSCHAP i have to use esp_wifi_sta_wpa2_ent_set_new_password, i tested both with same result(first 1 than 2 at the end 1&2 together)

```
I (1623) wifi: pp_task_hdl : 3ffc6e74, prio:23, stack:8192
I (1628) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (1633) wifi: mode : sta (24:0a:c4:03:fe:30)
D (1636) event: SYSTEM_EVENT_STA_START
I (1640) EH: 2
I (2515) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (3162) wifi: state: init -> auth (b0)
I (4162) wifi: state: auth -> init (2)
D (4162) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:NoobsLandWPAENT, ssid_len:15, bssid:84:26:15:42:29:9c, reason:2
```

```c
/* Hello World Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "ca.h"

#define USERNAME "anonymous@espressif.com"
#define PASSWORD "test123"

static EventGroupHandle_t wifi_event_group;

const int CONNECTED_BIT = BIT0;

static int evita_loop = 0;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    ESP_LOGI("EH","%d", event->event_id);
    switch(event->event_id) {
      case SYSTEM_EVENT_STA_START:
          //esp_wifi_connect();
          break;
      case SYSTEM_EVENT_STA_GOT_IP:
          xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
          break;
      case SYSTEM_EVENT_STA_DISCONNECTED:
          /* This is a workaround as ESP32 WiFi libs don't currently
             auto-reassociate. */
          if(evita_loop==0){
            evita_loop = 1;
            esp_wifi_connect();
          }
          xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
          break;
      default:
          break;
    }
    return ESP_OK;
}

void wifi_conn_init()
{
    tcpip_adapter_init();

    wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "NoobsLandWPAENT",
            .password = "",
            .bssid_set = false
        }
    };

    ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_enable() );
    esp_wifi_sta_wpa2_ent_clear_username();
    //esp_wifi_sta_wpa2_ent_clear_password();
    esp_wifi_sta_wpa2_ent_clear_new_password();
    esp_wifi_sta_wpa2_ent_clear_ca_cert();
    // TODO: set CA certificate
    esp_wifi_sta_wpa2_ent_set_username( (unsigned char *) USERNAME, strlen(USERNAME));
    //esp_wifi_sta_wpa2_ent_set_password( (unsigned char *) PASSWORD, strlen(PASSWORD));
    esp_wifi_sta_wpa2_ent_set_new_password( (unsigned char *) PASSWORD, strlen(PASSWORD));
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
    ESP_ERROR_CHECK( esp_wifi_connect() );

}

void app_main()
{
    nvs_flash_init();
    vTaskDelay(500 / portTICK_PERIOD_MS);
    wifi_conn_init();
}
```

joostd, 2017-01-10:
I suggest looking at the server logs. The client logs won't always tell you the real reason why authentication fails. Are you sure the `anonymous@espressif.com` account exists on your RADIUS server?

Gianpy86, 2017-01-10:
I can't check the log to the server because the server i'm using to develop is a https://www.ironwifi.com/ server. I'm pretty sure that the anonymous account exist, i tried from my mac book with that credential and it works correctly.

joostd, 2017-01-10:
Ah, I see. Maybe proxying through FreeRADIUS can help. See
https://github.com/joostd/esp8266-eduroam/wiki/Debugging

joostd, 2017-01-15:
Editing libwpa2 is no longer necessary: use

```
esp_err_t esp_wifi_sta_wpa2_ent_set_identity(unsigned char *identity, int len);
```
See
https://github.com/espressif/esp-idf/commit/b455b0246c09cc27589228b7222da8ada0a83f12

FayeY, 2017-05-03:
@Gianpy86 , is this problem still unsolved?

closed 2017-07-07 by FayeY