espressif/esp-idf #202: make monitor fails

shtirlic, 2017-01-07:
```
make monitor
Usage: miniterm.py [options] [port [baudrate]]

miniterm.py: error: no such option: --raw
make: *** [monitor] Error 2
```
```
uname -a
16.3.0 Darwin Kernel Version 16.3.0: Thu Nov 17 20:23:58 PST 2016; root:xnu-3789.31.2~1/RELEASE_X86_64 x86_64

```
```
python --version
Python 2.7.13

```

negativekelvin, 2017-01-07:
Maybe update pyserial?

seopyoon, 2017-01-11:
I navigated to the makefile, located at ```esp-idf/components/esptool_py/Makefile.projbuild``` and deleted ```--raw``` towards the very bottom and it works.  

```make monitor``` failure did not happen on my Mac, but happened on Ubuntu running on Virtualbox on Windows machine.

projectgus, 2017-01-11:
@seopyoon which version of Ubuntu, and which version of pyserial do you have? On Ubuntu, you should be able to check this with `dpkg -l python-serial`

cmason1978, 2017-01-11:
This is also failing for me and updating pyserial to 2.7 fixed it!  Thanks @negativekelvin !

projectgus, 2017-01-20:
Thanks for letting us know. The `--raw` option is necessary on newer pyserial otherwise the colour highlighting from log output is not rendered properly. However we should add a note to the README about this error and what causes it.

grisevg, 2017-01-26:
How to upgrade `python-serial` to 2.7 on Ubuntu 14.04?
Latest package in 14.04 repository is `pyserial 2.6-1build1`

projectgus, 2018-09-26:
This error should be taken care of now:

* "make monitor" will use idf_monitor.py not miniterm monitor, only "simple_monitor" uses miniterm.
* [requirements.txt file](https://github.com/espressif/esp-idf/blob/master/requirements.txt#L8) requires pyserial v3.0 or newer, which will support the `--raw` argument on miniterm.py.

Upgrading pyserial is the solution - either via pip, or via distribution package manager for Linux users.

closed 2018-09-26 by projectgus