espressif/esp-idf #204: make monitor doesn't work in MSYS2 32b, win7

michal037, 2017-01-07:
I using MSYS2 32bit on Windows 7, and pySerial version 3.2.1.
I wrote everything here: https://github.com/espressif/esp-idf/issues/200
I have got ESP32-devkitc with SILABS CP2102 on COM8 in my PC.

```
Michal@Michal MSYS /dev
$ ls
clipboard  conin  conout  console  dsp  fd  full  kmsg  mqueue  null  ptmx  pty0  random  scd0  sda  sda1  sda2  sda3  shm  sr0  stderr  stdin  stdout  tty  ttyS0  ttyS7  urandom  windows  zero

Michal@Michal MSYS /dev
$

```
MSYS2 see `COM8` as `ttyS7`, and so I set `ttyS7` in menuconfig.
For example: `COM1` is `ttyS0`.

When i write `make monitor` command in `esp-idf-template` project, make print this:

```
Michal@Michal MSYS /home/esp-idf-template
$ make monitor
don't know how to enumerate ttys on this system.
! I you know how the serial ports are named send this information to
! the author of this module:

sys.platform = 'msys'
os.name = 'posix'
pySerial version = 3.2.1

also add the naming scheme of the serial ports and with a bit luck you can get
this module running...
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "build/bdist.msys-2.6.0-i686/egg/serial/tools/miniterm.py", line 16, in <module>
  File "build/bdist.msys-2.6.0-i686/egg/serial/tools/list_ports.py", line 29, in <module>
  File "build/bdist.msys-2.6.0-i686/egg/serial/tools/list_ports_posix.py", line 96, in <module>
ImportError: Sorry: no implementation for your platform ('posix') available
make: *** [/home/esp-idf/components/esptool_py/Makefile.projbuild:76: monitor] Error 1
```

How to solve this problem ?

negativekelvin, 2017-01-07:
Use "COM8" in menuconfig

michal037, 2017-01-08:
@negativekelvin It doesn't work properly. I checked it out of curiosity. This same error.

projectgus, 2017-01-08:
`make monitor` will work if you follow the instructions for [Windows setup given here](http://esp-idf.readthedocs.io/en/latest/windows-setup.html). You don't need to use Cygwin /dev/ttyX paths, you can use COMx just fine.

For pyserial to work properly in MSYS2, the Python which is installed has to be a MinGW Python (which is Windows-native) rather than an MSYS/Cygwin Python. The instructions provided in #200 say to run "pacman -S python" which installs the MSYS Python. The `windows_install_prerequisites.sh` script that is part of the official instructions installs `mingw-w64-i686-python2-pip`, which pulls in mingw-w64-i686-python2 as a dependency.

To check which kind of Python you have, run `python -c "import os; print os.name"` in an MSYS terminal. `nt` means you have MinGW Python and pyserial will work as expected, `posix` means you have Cygwin Python and it will not work as expected.

closed 2017-01-08 by projectgus