espressif/esp-idf #205: Build stopped working:  No rule to make target 'libcoap/src/address.o', needed by 'libcoap.a'

michaelwgnr, 2017-01-09:
I just updated to 633cd49f881175ab5acf1b939e913b1d73a82329 and tried to build some of the examples (specifically, 01_hello_world and 14_gatts_server). Unfortunately, the project does not build anymore as a dependency can't be satisfied. The error message is the following:

```
make[1]: *** No rule to make target 'libcoap/src/address.o', needed by 'libcoap.a'.  Stop.
make: *** [/c/Daten/projekte/ESP32_eval/esp-idf/make/project.mk:378: coap-build] Error 2
```

Spritetm, 2017-01-09:
We added a new subproject; please do ``git submodule update --init`` to fetch the module on your local machine as well.

michaelwgnr, 2017-01-09:
Yes, that does it. Sorry to have bothered you. I'm not used to git submodules yet and did not use the `--init` flag.

closed 2017-01-09 by michaelwgnr