espressif/esp-idf #206: Can't configure GPIO 32 to 39 as "ext1 wakeup"

apolegato, 2017-01-09:
Hello,

When using:
 ```esp_err_t esp_deep_sleep_enable_ext1_wakeup(uint64_t mask, esp_ext1_wakeup_mode_t mode);```
to set a GPIO as External Wake Up, the documentation said:

> @param mask :  bit mask of GPIO numbers which will cause wakeup. Only GPIOs which are have RTC functionality can be used in this bit map: 0,2,4,12-15,25-27,32-39.

But according to the code of the function, it seems complicated to set GPIO 32 to 39 as External Wake Up:

```
esp_err_t esp_deep_sleep_enable_ext1_wakeup(uint64_t mask, esp_ext1_wakeup_mode_t mode)
{
    if (mode > ESP_EXT1_WAKEUP_ANY_HIGH) {
        return ESP_ERR_INVALID_ARG;
    }
    // Translate bit map of GPIO numbers into the bit map of RTC IO numbers
    uint32_t rtc_gpio_mask = 0;
    for (int  gpio = 0; mask; ++gpio, mask >>= 1) {
        if ((mask & 1) == 0) {
            continue;
        }
        if (!RTC_GPIO_IS_VALID_GPIO(gpio)) {
            ESP_LOGE(TAG, "Not an RTC IO: GPIO%d", gpio);
            return ESP_ERR_INVALID_ARG;
        }
        rtc_gpio_mask |= BIT(rtc_gpio_desc[gpio].rtc_num);
    }
    s_config.ext1_rtc_gpio_mask = rtc_gpio_mask;
    s_config.ext1_trigger_mode = mode;
    s_config.wakeup_triggers |= RTC_EXT_EVENT1_TRIG_EN;
    return ESP_OK;
}
```

For GPIO 32 to 39, the mask on the bit will exceed the size of the int, in the loop:
```for (int  gpio = 0; mask; ++gpio, mask >>= 1)```

negativekelvin, 2017-01-09:
Mask is 64bit, gpio just counts to 39.  Should be OK if mask = 1<<39?

apolegato, 2017-01-09:
Indeed..

Nevertheless I got this Warning:

>  warning: left shift count >= width of type 

When calling:
```esp_deep_sleep_enable_ext1_wakeup(BIT(GPIO_NUM_32), ESP_EXT1_WAKEUP_ANY_HIGH);```

igrr, 2017-01-09:
Could you please upload the full output of `make V=1`  into a gist and post a link here? Thanks.

apolegato, 2017-01-09:
Ok, it's working like this.

Thanks for the answer and sorry for pointing out a wrong issue.

closed 2017-01-09 by apolegato