espressif/esp-idf #208: Problem with uart example

ubxola, 2017-01-10:
Hello!
I have problems with lost data in the uart example.

In the sample app I have disabled the uart_echo_test and reconfigured the uart_evt_test to use UART0, configured RTS and CTS to suit my design and enabled uart flow control (.flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS).
Log tracing is redirected to UART1.

When I run the example data is lost and never echoed. However, if I uncomment the the log tracing in the uart_task all data is echoed and it seems to work ok. 

In my real application I see the same kind of problem with lost data when there are active tasks with higher priorities than the task reading the uart. This is my original problem and and very similar to what I see when testing the example.

I test by sending a file to the ESP32 using tera term in a windows environment and comparing the received echo with the original file. 
I have verified that RTS/CTS signalling works using an oscilloscope.

I have tried to reproduce the problem on the WROOM board but do not get flow control working on UART0.  In this setup I set RTS to IO15 and CTS to I013 which I got from the esp-wroom datasheet.
How can RTS/CTS flow control be tested on the WROOM board? 
Do I use the wrong pin configuration?

ubxola, 2017-01-10:
Update!
I missed that the uart is flushed on the BUFFER_FULL event. However if I remove the call to uart_flush() I still loose data and additionally get a crash.

forthlightning, 2017-01-10:
Did you get flow control working? I am also trying to use this functionality. 

Specifically I am tying to talk to a MODBUS device through a transceiver.

RX 15, TX 22, DriveEnable 5. In this case does it make sense to set one pin (5) for both RTS and CTS? These are physically tied together on the hardware side.

ubxola, 2017-01-11:
I never got RTS/CTS working on WROOM. I found a configuration that worked for my project. With less trace logs and txbuffer disabled the example worked well in my test setup.

closed 2017-01-11 by ubxola