espressif/esp-idf #210: When might SPI client mode be supported?

collin80, 2017-01-10:
It's awesome that a few days ago the SPI master driver was uploaded. However, I really need to be able to use the ESP32 in client mode on SPI. The old ESP8266 chip has a client driver but not this newer one. I'd try to do it myself but the SPI interface is not well documented at this point. Are there any plans to implement client mode in the near future?

bvernoux, 2017-01-10:
We are all waiting for SPI support (master/slave) and it is coming see latest post http://www.esp32.com/viewtopic.php?f=10&t=635&start=30

FayeY, 2017-05-02:
Hi , SPI slave has been implemented in [46fa2cf](https://github.com/espressif/esp-idf/commit/46fa2cfb4680b027a93c07c8616a4aa6e76d06ab)

closed 2017-05-17 by Spritetm