espressif/esp-idf #213: SPI master and clock speeds ... breaks down with speeds > 75KHz

nkolban, 2017-01-10:
I was experimenting with SPI speeds by setting the clock_speed_hz when calling `spi_bus_add_device`.  I thought I'd start at 100KHz so supplied a value of 100000 but when I placed a logic analyzer on the clock, I found that the speed was in excess of 2.8MHz ... examining the sample, I found that the sample uses 8KHz so I tried that and got a perfect clock cycle of 8KHz.  I worked my way up and found that at 75KHz, all is still well ... however, when we approach 80KHz, things start breaking down ... and at 80KHz, there is no obvious clock to be found.

This issue asks for an examination/comment.  I can't easily provide a recreate as my environment is a very large and complex application and difficult to separate out the SPI calls however I'm hoping that the puzzle will "simply" manifest itself if you try with a clock speed of 80KHz in your own tests.

negativekelvin, 2017-01-11:
https://github.com/espressif/esp-idf/blob/master/components/driver/spi_master.c#L413

Looks like bug in self-described 

> This algo could use some tweaking

Prescaler calculation should not divide by 32 then compare to 32?

Spritetm, 2017-01-11:
Hum, guess I'll just copy-paste the algo from the Arduino driver then.

negativekelvin, 2017-01-11:
Sounds good.  Also wasn't sure about

> //In hw, n, h and l are 1-32 

when they are 6-bit values

Spritetm, 2017-01-11:
Ah, good catch. I can't read datasheets, seemingly. Which is somewhat worrying, seeing as I help writing them... Anyway, I'll fix that as well.

nkolban, 2017-01-13:
Howdy ... a procedural question ... should I assume that until this issue is updated or closed by y'all that the issue still exists?  Specifically, I'm curious as to what should be a flag to me that I should re-test and/or return to attempting to use the library without restrictions.

igrr, 2017-01-15:
This has been fixed in 8a2ff5a — @Spritetm forgot to #mention this issue in the commit.

closed 2017-01-15 by igrr