espressif/esp-idf #219: "waiti" called and triggers

nathanjel, 2017-01-12:
Hi,

I have a complex application (MQTT client/server, UART sending and receiving messages every 10ms, SPI Flash access from within the program and of course WIFI) and have noticed a strange random scenario that manifests as below. I use both CPU cores in my application.

It is quite strange to notice this. Seems like "waiti" (wait for interrupt) gets called to wait for interrupt, and as that does not happen, watchdog triggers. One would expect at least the FreeRTOS tick to fire.

Also this happens when I trigger some network IO, e.g. connect to the softAP on the device. I do not know how to research this further, and even whether the issue is in my app or within the framework itself. Any guidance/ideas highly welcomed.

I have seen this issue also with a different stack trace, e.g. including the uart_rx_*** interrupt handler from the esp-idf device component.

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x400d280c  PS      : 0x00060234  A0      : 0x8008508c  A1      : 0x3ffbc770  
A2      : 0x00000008  A3      : 0x3ffb307c  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffb5f24  A8      : 0x3ffb3098  A9      : 0x3ffbc750  
A10     : 0x00000000  A11     : 0x00060223  A12     : 0x00060223  A13     : 0xb33f0000  
A14     : 0xb33fffff  A15     : 0x00060223  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d280c:0x3ffbc770 0x4008508c:0x3ffbc790

And that traces in GDB to 

0x400d280c in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
52	        asm("waiti 0");
(gdb) bt
#0  0x400d280c in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
#1  0x4008508c in prvIdleTask (pvParameters=0x0) at /home/nathan/xnode/esp-idf/components/freertos/./tasks.c:3293
(gdb)

negativekelvin, 2017-01-12:
Does your code disable interrupts?  Have you tried increasing the timeout period?

nathanjel, 2017-01-12:
My code (read: the code I have written) does not enable or disable interrupts specifically. I written my code using FreeRTOS threads. However, the specific library functions (UART, SPI) do disable/enable interrupts.

I have not tried to increase the timeout period. It is standard 300 ms. Given FreeRTOS ticks every 10ms, that should be more than enough, shouldn't it?

nathanjel, 2017-01-12:
Maybe the case is related to interrupt priority? I get a different "version" of similar issue - again interrupt watchdog (this time on CPU0) on "waiti" which looks like being called inside of another interrupt (this time in UART interrupt).

Below results on most recent ESP-IDF master (12th Jan) and the newest tools package.

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x400d2808  PS      : 0x00060034  A0      : 0x80083194  A1      : 0x3ffb0c10  
A2      : 0x00000001  A3      : 0x00000001  A4      : 0x0000001d  A5      : 0x3ffb0c38  
A6      : 0x40086238  A7      : 0x3ffd22cc  A8      : 0x8008423d  A9      : 0x3ffb0c00  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00060021  
A14     : 0x3ffb1368  A15     : 0x00060820  SAR     : 0x00000018  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d2808:0x3ffb0c10 0x40083194:0x3ffb0c30 0x400815e9:0x3ffb0c70

0x400d2808 in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
52	        asm("waiti 0");
(gdb) p
The history is empty.
(gdb) bt
#0  0x400d2808 in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
#1  0x40083194 in uart_rx_intr_handler_default (param=0x3ffd2020)
    at /home/nathan/xnode/esp-idf/components/driver/./uart.c:491
#2  0x400815e9 in _xt_lowint1 ()
(gdb)

nathanjel, 2017-01-13:
Please, any hints for further research are welcomed. I don't know ESP32 internals too well (altough I got my hands on some meaty PDF from Xtensa), but happy to dig deeper, given any hints on the reasons from these more involved.

nathanjel, 2017-01-15:
I have worked around this issue by changing gettimeofday menuconfig settings to use only FRC1 timer for time relevance. The issue was occuring when it was set to RTC only. I have not tested with FRC1&RTC enabled at the same time. Looking at the code, I believe there is an issue if get_rtc_time_us() is entered twice, but only one of them exits and the other locks.

This code causes the problem in time.c
```
static uint64_t get_rtc_time_us()
{
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE_M);
    while (GET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_VALID_M) == 0) {
        ;
    }
    CLEAR_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE_M);
```

Consider enabling re-entrancy in get_rtc_time_us(), get_time_since_boot() or _gettimeofday_r() in time.c to avoid this issue.

igrr, 2017-01-15:
Thanks for reporting this. Do you have some example code to reproduce the issue?

nathanjel, 2017-01-15:
I would need to share with You my whole big commercial project. I don't have a singled out test case, the way I test it is I start the device, enable load on controls and network and wait for fail. Changing menuconfig to FRC1 only removed the above Guru Medidations, otherwise they would occur within several minutes from boot.

My code calls gettimeofday frequently in several parallel threads, and I have seen the function name popup randomly, that lead me to investigate the function more. If (and that is not a promise) I have more time, I am happy to try solving this and repeating tests on my code-base.

igrr, 2017-01-16:
I took a shot at writing a test case for this. This test case runs 4 tasks which call gettimeofday, 2 tasks on each CPU. Seemingly this is not sufficient to reproduce the issue:

```c++
TEST_CASE("gettimeofday can run in many tasks in parallel", "[newlib]")
{
    void gettimeofday_task(void* arg)
    {
        for (int i = 0; i < 200000; ++i) {
            struct timeval tv;
            gettimeofday(&tv, NULL);
        }
        SemaphoreHandle_t * p_done = (SemaphoreHandle_t *) arg;
        xSemaphoreGive(*p_done);
        vTaskDelay(1);
        vTaskDelete(NULL);
    }

    SemaphoreHandle_t done = xSemaphoreCreateCounting(4, 0);

    xTaskCreatePinnedToCore(&gettimeofday_task, "gettimeofday", 4096, &done, 5, NULL, 0);
    xTaskCreatePinnedToCore(&gettimeofday_task, "gettimeofday", 4096, &done, 5, NULL, 0);
    xTaskCreatePinnedToCore(&gettimeofday_task, "gettimeofday", 4096, &done, 5, NULL, 1);
    xTaskCreatePinnedToCore(&gettimeofday_task, "gettimeofday", 4096, &done, 5, NULL, 1);

    for (int i = 0; i < 4; ++i) {
        xSemaphoreTake(done, 10000/portTICK_PERIOD_MS);
    }
    // no specific tests here — this should not crash
}
```

So there's probably more to this issue than just gettimeofday.
Let me know if you have any clues what else should be added to the test in order to reproduce the issue.

negativekelvin, 2017-01-17:
@igrr No locks in get_rtc_time_us() ?
https://github.com/espressif/esp-idf/blob/ee9fb10e29adfcd74817a551f9343f77d58f703f/components/newlib/time.c#L45

igrr, 2017-01-17:
Good point — there's no lock there, and it *may* cause a problem if two concurrent gettimeofday calls happen when the high word of RTC counter changes. But this is not something that can cause interrupt watchdog timeout — merely one of the readings will be incorrect.

negativekelvin, 2017-01-17:
@igrr well I was thinking if update bit is cleared while other call is in while loop waiting for update it could get stuck but maybe not possible

nathanjel, 2017-01-18:
igrr, i have experimented off Your test case template, adding queues and uart rx driver and it seems my initial thoughts were wrong, however it remains a fact that removal of RTC reference made a huge difference on stability of our code.

exploring this issue further, I have found that this issue will occur with specific settings of interrupt allocator flags for UART interrupt, and not with others e.g.

good: `uart_driver_install(PC_UART_NUM_IN_ESP32, PC_UART_IN_BUFF_SIZE, 0, 0, NULL, ESP_INTR_FLAG_LEVEL3 | ESP_INTR_FLAG_IRAM);` - code runs much longer, and eventually I will get this dump: #242 

bad: `uart_driver_install(PC_UART_NUM_IN_ESP32, PC_UART_IN_BUFF_SIZE, 0, 0, NULL, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM);` - i get the issue as described above

However it's all reading code and Xtensa PDF, trial and error, while trying to meet the deadlines. But it seems there must be something around these interrupt levels, e.g. see my pull request #256 which addresses a different issue.

nathanjel, 2017-01-25:
This is definitely not closed. On the newest esp-idf

0x400d3200 in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
52	        asm("waiti 0");
(gdb) bt
#0  0x400d3200 in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
#1  0x40085573 in xTaskIncrementTick () at /home/nathan/xnode/esp-idf/components/freertos/./tasks.c:2408
#2  0x4008456e in xPortSysTickHandler () at /home/nathan/xnode/esp-idf/components/freertos/./port.c:225
#3  0x40086264 in _frxt_timer_int ()
#4  0x40081691 in _xt_lowint1 ()
(gdb)

nathanjel, 2017-01-26:
Thanks!

In the meantime I resolved (or worked around, You tell me ;-)) this problem by applying this change 

```
diff --git a/components/freertos/xtensa_vectors.S b/components/freertos/xtensa_vectors.S
index 7baae07..68ddeae 100644
--- a/components/freertos/xtensa_vectors.S
+++ b/components/freertos/xtensa_vectors.S
@@ -647,7 +647,7 @@ _xt_user_exc:
     #if XCHAL_HAVE_WINDOWED
     s32e    a0, sp, -12                     /* for debug backtrace */
     #endif
-    rsr     a0, PS                          /* save interruptee's PS */
+    rsil    a0, 0x7                         /* save interruptee's PS, block interrupts */
     s32i    a0, sp, XT_STK_PS
     rsr     a0, EPC_1                       /* save interruptee's PC */
     s32i    a0, sp, XT_STK_PC
@@ -1080,7 +1080,7 @@ _xt_lowint1:
     mov     a0, sp                          /* sp == a1 */
     addi    sp, sp, -XT_STK_FRMSZ           /* allocate interrupt stack frame */
     s32i    a0, sp, XT_STK_A1               /* save pre-interrupt SP */
-    rsr     a0, PS                          /* save interruptee's PS */
+    rsil    a0, 0x7                         /* save interruptee's PS, block interrupts */
     s32i    a0, sp, XT_STK_PS
     rsr     a0, EPC_1                       /* save interruptee's PC */
     s32i    a0, sp, XT_STK_PC

```

nathanjel, 2017-01-28:
Another example

```
Remote debugging using /dev/ttyUSB1
0x400d2b60 in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
52	        asm("waiti 0");
(gdb) bt
#0  0x400d2b60 in esp_vApplicationIdleHook () at /home/nathan/xnode/esp-idf/components/esp32/./freertos_hooks.c:52
#1  0x4008309f in uart_rx_intr_handler_default (param=0x3ffda044)
    at /home/nathan/xnode/esp-idf/components/driver/./uart.c:491
#2  0x400815c5 in _xt_medint3 ()
(gdb) 
```

nathanjel, 2017-02-14:
Can't repeat this dilligently within newest esp-idf.

nathanjel, 2017-02-15:
After playing some more time, there actually seems to be a theme - i would get it most likely with LEVEL2 and LEVEL3 interrupts, crashing somewhere on entering or exiting critical sections (e.g. `UART_EXIT_CRITICAL_ISR` or `taskENTER_CRITICAL_ISR`). Could it be a cross-core deadlock issue, e.g. if we get one CPU in a critical section from task code and and then interrupt on the other CPU entering critical section using same mutex?

FayeY, 2017-05-03:
@nathanjel Is this problem still unsolved now?

closed 2017-11-28 by FayeY