espressif/esp-idf #220: Examples fail build, returning: "cannont find -lrtc"

StanTroeh, 2017-01-12:
**Do a clean fetch of the esp-idf tree from MSYS32 console, move to example directory & build**
Stan@Stan-PC MSYS /c/Espressif/ESP32 $ **/usr/bin/git clone https://github.com/espressif/esp-idf.git esp-idf**
Cloning into 'esp-idf'...
remote: Counting objects: 16091, done.
remote: Total 16091 (delta 0), reused 0 (delta 0), pack-reused 16091
Receiving objects: 100% (16091/16091), 13.05 MiB | 395.00 KiB/s, done.
Resolving deltas: 100% (10510/10510), done.
Checking connectivity... done.
Checking out files: 100% (1812/1812), done.

Stan@Stan-PC MSYS /c/Espressif/ESP32/esp-idf/examples/01_hello_world $ **echo $IDF_PATH**
/c/Espressif/ESP32/esp-idf
Stan@Stan-PC MSYS /c/Espressif/ESP32 $ **cd esp-idf/examples/01_hello_world/**
Stan@Stan-PC MSYS /c/Espressif/ESP32/esp-idf/examples/01_hello_world $ **make**
make[1]: Entering directory '/c/Espressif/ESP32/esp-idf/tools/kconfig'
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -lncursesw -lintl
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncursesw -lintl
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncursesw -lintl
make[1]: Leaving directory '/c/Espressif/ESP32/esp-idf/tools/kconfig'
DEFCONFIG
\#
\# configuration written to /c/Espressif/ESP32/esp-idf/examples/01_hello_world/sdkconfig
\#
MENUCONFIG
     **took defaults, selected EXIT**

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

GENCONFIG
WARNING: Missing submodule components/micro-ecc/micro-ecc...
Attempting 'git submodule update --init components/micro-ecc/micro-ecc' in esp-idf root directory...
Submodule 'components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/micro-ecc/micro-ecc'
Cloning into '/c/Espressif/ESP32/esp-idf/components/micro-ecc/micro-ecc'...
': not a valid identifierline 89: export: `sha1
': not a valid identifierline 89: export: `displaypath
Submodule path '': checked out '14222e062d77f45321676e813d9525f32a88e8fa'
WARNING: Missing submodule components/esptool_py/esptool...
Attempting 'git submodule update --init components/esptool_py/esptool' in esp-idf root directory...
Submodule 'components/esptool_py/esptool' (https://github.com/themadinventor/esptool.git) registered for path 'components/esptool_py/esptool'
Cloning into '/c/Espressif/ESP32/esp-idf/components/esptool_py/esptool'...
': not a valid identifierline 89: export: `sha1
': not a valid identifierline 89: export: `displaypath
Submodule path '': checked out 'fe69994270e2a450aad3e94a409b58460b1a214f'
CC src/bootloader_random.o
CC src/flash_encrypt.o
CC src/esp_image_format.o
CC src/flash_partitions.o
CC src/secure_boot_signatures.o
CC src/secure_boot.o
CC src/efuse.o
CC src/bootloader_flash.o
AR libbootloader_support.a
CC log.o
AR liblog.a
CC spi_flash_rom_patch.o
AR libspi_flash.a
CC micro-ecc/uECC.o
AR libmicro-ecc.a
CC bootloader_start.o
AR libmain.a
LD bootloader.elf
**c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lrtc
collect2.exe: error: ld returned 1 exit status**
make[1]: *** [/c/Espressif/ESP32/esp-idf/make/project.mk:310: /c/Espressif/ESP32/esp-idf/examples/01_hello_world/build/bootloader/bootloader.elf] Error 1
make: *** [/c/Espressif/ESP32/esp-idf/components/bootloader/Makefile.projbuild:31: /c/Espressif/ESP32/esp-idf/examples/01_hello_world/build/bootloader/bootloader.bin] Error 2

**Similar results on example 02_blink and 03_http_request**

I did a search for reference to the library:
$ **grep -r "\-lrtc"** 
**build/bootloader/main/component_project_vars.mk**:COMPONENT_LDFLAGS += -L $(IDF_PATH)/components/bootloader/src/main -lmain -T esp32.bootloader.ld -T $(IDF_PATH)/components/esp32/ld/esp32.rom.ld -T esp32.bootloader.rom.ld -L $(IDF_PATH)/components/esp32/lib/ **-lrtc**
**build/esp32/component_project_vars.mk**:COMPONENT_LDFLAGS += -lesp32 $(IDF_PATH)/components/esp32/libhal.a -L$(IDF_PATH)/components/esp32/lib -lcore -lnet80211 -lphy **-lrtc** -lpp -lwpa -lsmartconfig -lcoexist -lwps -lwpa2 -L $(IDF_PATH)/components/esp32/ld -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld

ry111, 2017-01-23:
I am started having the same issue ("cannot find -lrtc") once I updated esp-idf today. Any suggestions?

Thanks

krzychb, 2017-01-23:
Hi @StanTroeh,

Basing on the log I believe you are missing `--recursive` in the first command. 

> Stan@Stan-PC MSYS /c/Espressif/ESP32 $ /usr/bin/git clone https://github.com/espressif/esp-idf.git esp-idf

Instead it should be:

```
git clone --recursive https://github.com/espressif/esp-idf.git
```
See additional description in http://esp-idf.readthedocs.io/en/latest/linux-setup.html#step-2-getting-esp-idf-from-github. It is applicable to Windows installation as well.

projectgus, 2017-01-25:
Hi @StanTroeh,

I think @krzychb is right here. We have warnings in esp-idf that should be printed if submodules are missing, but I think there isn't one for this bootloader step. We'll fix this so the error is easier to diagnose.

If you don't want to re-clone, running `git submodule update --init --recursive` should fix an existing working directory that's seeing this problem.

Angus

StanTroeh, 2017-01-26:
I eventually deleted the whole collection and then ran the git again, this time with --recursive.  Another problem pointed to an execution issue with python 2.7.  I'm uncertain if the resolution was the 'git' (likely) or the resolving PYTHONHOME (once upgrading from 2.5 to 2.7) that finally resolved the problem.  Thanks for the help!

closed 2017-02-22 by igrr

benjaminaigner, 2017-03-08:
I had the same problem, `git submodule update --init --recursive` fixed it for me.

Thanks!

jcabad100, 2019-05-18:
I did clone the the github (it was clearly stated that an unzip would not work), nevertheless i have the same problem and the "git submodule update --init --recursive" doesn't seems to retrieve anything new....all the *.o files are properly built on first make attempt but the linker phase with followin message (2nd run): "DELL@DESKTOP-MVQO1NB MINGW32 ~/esp/esp-idf/examples/get-started/hello_world
$ make
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from D:/IoT/Espressif/Toolchain/msys32/home/DELL/esp/esp-idf/requirements.txt are satisfied.

App "hello-world" version: v4.0-dev-546-gad3b820e7
LD build/hello-world.elf
d:/iot/espressif/toolchain/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -ljsmn
collect2.exe: error: ld returned 1 exit status
make: *** [D:/IoT/Espressif/Toolchain/msys32/home/DELL/esp/esp-idf/make/project.mk:521: /home/DELL/esp/esp-idf/examples/get-started/hello_world/build/hello-world.elf] Error 1
" ...any hint or idea please? ...thank you!

jcabad100, 2019-05-19:
Today i tried again running "git submodule update --init --recursive" and despite there is no verbose/feedback and also no error after some seconds something should have happened. Today i retried the make command (just make , not make flash as my ESP32 WROVER KIT should arrive tomorrow)....apparently all ready now...