espressif/esp-idf #222: Bug: 04-https_request problem (mbedtls handshake) after several calls

ivanO86, 2017-01-12:
Hi,

I am working in an application which ask for some information each 5 min (aprox) via https. After several calls (maybe after 3 hours), the SSL handshake starts to fail. I took a lot of time debugging my code without any result (bad crl or crt init or something like that), so I tried your example, getting the same error:

E (57867091) example: mbedtls_ssl_handshake returned -0x4290
E (57867101) example: Last error was: -0x4290 - RSA - The public key operation failed : BIGNUM - Memory allocation failed

or

E (57808461) example: mbedtls_ssl_handshake returned -0x4e
E (57808461) example: Last error was: -0x4e - UNKNOWN ERROR CODE (004E)

My code is a bit different because I create and free all resources each call, but I get the same code.

Best regards.

ivanO86, 2017-01-27:
It seems to be a memory leak into the mbedtls library. I am using xPortGetFreeHeapSize() to check the heap size and with each ssl communication the free heap size is lower.

projectgus, 2017-04-20:
Hi,

Sorry for the extended delay getting back to you. I actually tried to reproduce this a couple of months ago, and I obviously never got around to replying with a comment - entirely my fault, and I apologise.

I'm not able to reproduce this, at least with the HTTPS example. If I print the free heap each time around the loop, then I see it steadily go down for the first minute or so. This is relatively normal, and due to TCP sockets persisting in the TIME_WAIT state until LWIP can delete them. After the first couple of minutes, the heap usage stabilises (sockets are being created at the same rate they are timing out). From that point, the heap usage stays constant each time around the loop.

I have left the HTTPS example running in this way for upwards for around a day, no further problems are seen.

Is there anything that you changed in the HTTPS example code or config that might be triggering the memory leak?

Angus

ivanO86, 2017-04-21:
Hello Angus,

I tried it with the v1.0 release. I am working in another projects and I couldn't test it with the last release.

The only change I made was force the certificate check, instead of  the MBEDTLS_SSL_VERIFY_OPTIONAL flag, but checking the code I don't think so this can be the reason for the error.

If you cannot reproduce this with the latest release I suppose the bug is fixed :). Thanks a lot.

BR

closed 2017-05-03 by FayeY