espressif/esp-idf #225: ESP32 gap scan issue

alejandroviegener, 2017-01-12:
I have been experimenting with the "ESP32 Thing" module from Sparkfun. I installed the toolchain, the IDF and reproduced some of the examples (GPIO, I2C, Wifi http request...)

However, I am having a problem using the BLE in scanning mode. I loaded the 15_gatt_client example, opened the console to see the Logs, but I never receive any advertising reports, despite there being advertising BT devices around. 

I actually only need to see the advertising reports from nearby BT devices, no connection is requiered. Is there any example available to test this?

I also  modified  the gatt_client_example, in the following way, eliminating all the gatt module functionality, without success...

`
//////////////////////////////////////////////////////////////////////////////////////////////////////////

void gattc_client_test(void)
{
    esp_bluedroid_init();
    esp_bluedroid_enable();
    //ble_client_appRegister();    
    
    esp_err_t status;
    ESP_LOGI( GATTC_TAG, "register callback\n" );

    //register the scan callback function to the gap moudule
    if ((status = esp_ble_gap_register_callback(esp_gap_cb)) != ESP_OK) 
    {
        ESP_LOGE(GATTC_TAG, "gap register error, error code = %x\n", status);
        return;
    }
    
    // Set scan parameters and star scanning ------
    esp_ble_gap_set_scan_params( &ble_scan_params );
}
void app_main()
{
    esp_bt_controller_init();
    gattc_client_test();
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////
`
Am I missing something?

nkolban, 2017-01-20:
In case it might be of use, here is a scanner for ESP32 that I have been using:

https://github.com/nkolban/esp32-snippets/tree/master/BLE/scanner

closed 2017-01-20 by alejandroviegener

neilyoung, 2017-03-02:
OT: Is there any technical specification known regarding the question: How many iBeacons can be seen simultaneously per time frame?

neilyoung, 2017-03-12:
@nkolban: The code is not working anymore w/o modifications, especially in initialization code.