espressif/esp-idf #226: Warnings logged on build that toolchain version is not supported

nkolban, 2017-01-13:
Am using ESP-IDF as of the date of this issue.  I got a warning that my toolchain was not supported.  Great that there is a check.  However, after installing the latest toolchain, I continue to get the same error.  Here is a text capture of what I am seeing:
```
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ make
WARNING: Toolchain version is not supported: 
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
/home/kolban/esp32/esptest/esp-idf/make/project.mk:445: WARNING: Compiler version is not supported: 
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
```

As you can see, the execution of my `xtensa-esp32-elf-gcc` is producing the version that appears to be expected but yet the error is still logged.

negativekelvin, 2017-01-13:
You may want to check the output of the actual commands on your system since your output seems to show null for both items

https://github.com/espressif/esp-idf/blob/master/make/project.mk#L432

projectgus, 2017-01-13:
This is on Linux, yes? I remember Ubuntu, but which version?

Despite the warning, does the build run correctly? There is a bug that if the path to the toolchain needs to be set in menuconfig, it will print a warning like this before the path is configured. However it looks like the toolchain is already on your PATH, so it may be something else in this case.

projectgus, 2017-01-13:
@nkolban  do you only get this the first time you run 'make'?

MartyMacGyver, 2017-01-13:
This isn't working right for me either - and now I see why. I'm racing out the door but here's the fix:

    TOOLCHAIN_COMMIT_DESC := $(shell $(CC) --version | sed -E -n 's|xtensa-esp32-elf-gcc.*\ \(([^)]*).*|\1|gp')
    TOOLCHAIN_GCC_VER := $(shell $(CC) --version | sed -E -n 's|xtensa-esp32-elf-gcc.*\ \(.*\)\ (.*)|\1|gp')

Why? Because if you're using Windows executables it reports with .exe:

$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

projectgus, 2017-01-13:
Thanks @MartyMacGyver . We'll have a fix for Windows version detection out shortly.

nkolban, 2017-01-13:
@projectgus 

The make appears clean ... i.e. it runs to complete.

If I then run make clean and make again, the warnings are still present.

projectgus, 2017-01-13:
OK, thanks @nkolban for the update. We have a fix for both these issues (warning on a clean build, warning on Windows) which will be merged shortly.

MartyMacGyver, 2017-01-13:
The fix should be one and the same.

projectgus, 2017-01-13:
They're not, but they're similar.

Frida854, 2017-01-15:
I have boiled it down to on linux
export CONFIG_TOOLPREFIX=$(pwd)/xtensa-esp32-elf/bin/xtensa-esp32-elf-
before I ran 
make duktape_configure and
make
then it sorted it out for me.

projectgus, 2017-01-15:
Thanks for that extra detail @Frida854 .

There's a fix in the pipeline for this warning, which is two different underlying issues depending on platform:
* Windows doesn't detect toolchain correctly
* If the toolchain needs to be set in menuconfig before it works (ie not in PATH, or has a different prefix to the default prefix), then the warning appears on a clean build (until after "menuconfig" or "defconfig") has run.)

commit 5c9c08ea (2017-01-16): Toolchain detection: Fix issue when run in a clean project

If the makefile config entry hasn't been generated yet, don't test the toolchain.

Closes #226 https://github.com/espressif/esp-idf/issues/226
closed 2017-01-16 by igrr