espressif/esp-idf #227: sdmmc_card_init: send_scr returned 0x109

mtmsweb, 2017-01-13:
I test esp on self-made PCB, some connection problem are possible. 
Only tested card, that I have now, is 2GB SanDisk (China):

sdmmc_card_init result in **sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109**. 
csd and cid read correctly: 

> sd_card: csd_ver : 0
> sd_card: mmc_ver : 0
> sd_card: capacity : 3970048
> sd_card: sector_size : 512
> sd_card: mfg_id : 3
> sd_card: oem_id : 21316
> sd_card: name : SD02G

igrr, 2017-01-13:
0x109 is `ESP_ERR_INVALID_CRC`, and `send_scr` is the first transaction which uses D0 line (all the previous ones use only CMD and CLK). Suggest checking what happens on D0 line (that's GPIO2) using an oscilloscope. You may be missing a pull up resistor on this line, or you may have both a pull down resistor and a pull up one.

Note that flashing the ESP with a pull up resistor attached to GPIO2 will not work, because GPIO2 needs to be low when going into the bootloader mode. One solution is to connect GPIO2 and GPIO0 with a jumper — this will pull down GPIO2 along with GPIO0 during programming.

closed 2017-01-13 by igrr

vinifr, 2018-07-20:
Sorry for post here, I have same problem, but i am getting a different value. I am getting this:
_sdmmc_cmd: sdmmc_card_init: send_scr (2) returned 0xffffffff_

Could be the same problem?

tschundler, 2019-06-07:
For anyone else landing here from searching for this 0x109 error, my solution was simply telling it to use the pullups in software -
For Arduino:
```
  pinMode(15, INPUT_PULLUP);
  pinMode(2, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13, INPUT_PULLUP);
```
or ESP-IDF
```
    gpio_set_pull_mode(15, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(2, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(4, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(12, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(13, GPIO_PULLUP_ONLY);
```

puyams, 2020-06-19:
Work...!
Thanks

EdBerde, 2021-03-07:
I have found that in Arduino Studio, when using the Camera and SD_MMC together on the ESP32 CAM board, if I do **anything** with pin 4 **anywhere** in the code, the camera will sporadically fail (error messages listed below).

I have absolutely verified this many times now that I have tracked it down, as this has plagued me for weeks.

If I have the following line anywhere in my code, even inside an if statement where it is not being called, my camera will sometimes fail when I am using SD:
pinMode(4, INPUT_PULLUP);

Below is the set of error messages I sporadically get when I include **any** pinMode for GPIO 4:
[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
Camera capture failed

Again, this is specific only to the case where you are using SD_MMC.h and the Camera together on an ESP32-CAM board.  It may also only be limited to Arduino Studio.

It is my belief that by including pinMode anywhere in your code, either Arduino or the ESP32 configures that pin for IO, which causes issues with the SD_MMC and the camera's SCCB interface.  There were other posts I found where it said that you had to enable SD_MMC by calling SD_MMC.begin() prior to initializing the camera, or there would be issues with the registration on the SCCB interface (or something like that).  I now call SD_MMC.begin() right at the beginning of my setup() function and I do not use GPIO4 for anything - and of course I don't do any pinMode on GPIO4 anymore!

Hope this helps someone!