espressif/esp-idf #228: Exception causes looping when "enter gdb stub now" selected

nkolban, 2017-01-13:
I encountered an exception in my logic so changed my configuration to ensure GDB stub.  Now I find that instead of entering the GDB ready environment as previous builds did, instead we go into a loop where the following loops :

```
Entering gdb stub now.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d0cf0  PS      : 0x00060033  A0      : 0x80085e46  A1      : 0x3ffb8800  
A2      : 0x3ffb88a0  A3      : 0x00000000  A4      : 0x3ffb2334  A5      : 0x383fc000  
A6      : 0x3ff60000  A7      : 0x00000000  A8      : 0x80085800  A9      : 0x0000007d  
A10     : 0x3ffb88a0  A11     : 0x00000000  A12     : 0x3ff40000  A13     : 0x00000000  
A14     : 0x00000030  A15     : 0x3ffb20bc  SAR     : 0x00000014  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d0cf0:0x3ffb8800 0x40085e46:0x3ffb8880 0x400810b5:0x3ffb88a0 0x400d0cf0:0x3ffb8960 0x400d0cf0:0x3ffb89e0 0x400810b5:0x3ffb8a00 0x400d0cf0:0x3ffb8ac0 0x400d0cf0:0x3ffb8b40 0x400810b5:0x3ffb8b60 0x400d0cf0:0x3ffb8c20 0x400d0cf0:0x3ffb8ca0 0x400810b5:0x3ffb8cc0 0x400d0cf0:0x3ffb8d80 0x400d0cf0:0x3ffb8e00 0x400810b5:0x3ffb8e20 0x400d0cf0:0x3ffb8ee0 0x400d0cf0:0x3ffb8f60 0x400810b5:0x3ffb8f80 0x400d0cf0:0x3ffb9040 0x400d0cf0:0x3ffb90c0 0x400810b5:0x3ffb90e0 0x400d0cf0:0x3ffb91a0 0x400d0cf0:0x3ffb9220 0x400810b5:0x3ffb9240 0x400d0cf0:0x3ffb9300 0x400d0cf0:0x3ffb9380 0x400810b5:0x3ffb93a0 0x400d0cf0:0x3ffb9460 0x400d0cf0:0x3ffb94e0 0x400810b5:0x3ffb9500 0x400d0cf0:0x3ffb95c0 0x400d0cf0:0x3ffb9640 0x400810b5:0x3ffb9660 0x400d0cf0:0x3ffb9720 0x400d0cf0:0x3ffb97a0 0x400810b5:0x3ffb97c0 0x400d0cf0:0x3ffb9880 0x400d0cf0:0x3ffb9900 0x400810b5:0x3ffb9920 0x400d0cf0:0x3ffb99e0 0x400d0cf0:0x3ffb9a60 0x400810b5:0x3ffb9a80 0x400d0cf0:0x3ffb9b40 0x400d0cf0:0x3ffb9bc0 0x400810b5:0x3ffb9be0 0x400d0cf0:0x3ffb9ca0 0x400d0cf0:0x3ffb9d20 0x400810b5:0x3ffb9d40 0x400d0cf0:0x3ffb9e00 0x400d0cf0:0x3ffb9e80 0x400810b5:0x3ffb9ea0

Entering gdb stub now.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d0cf0  PS      : 0x00060033  A0      : 0x80085e46  A1      : 0x3ffb86a0  
A2      : 0x3ffb8740  A3      : 0x00000000  A4      : 0x3ffb2334  A5      : 0x383fc000  
A6      : 0x3ff60000  A7      : 0x00000000  A8      : 0x80085800  A9      : 0x0000007d  
A10     : 0x3ffb8740  A11     : 0x00000000  A12     : 0x3ff40000  A13     : 0x00000000  
A14     : 0x00000030  A15     : 0x3ffb20bc  SAR     : 0x00000011  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d0cf0:0x3ffb86a0 0x40085e46:0x3ffb8720 0x400810b5:0x3ffb8740 0x400d0cf0:0x3ffb8800 0x400d0cf0:0x3ffb8880 0x400810b5:0x3ffb88a0 0x400d0cf0:0x3ffb8960 0x400d0cf0:0x3ffb89e0 0x400810b5:0x3ffb8a00 0x400d0cf0:0x3ffb8ac0 0x400d0cf0:0x3ffb8b40 0x400810b5:0x3ffb8b60 0x400d0cf0:0x3ffb8c20 0x400d0cf0:0x3ffb8ca0 0x400810b5:0x3ffb8cc0 0x400d0cf0:0x3ffb8d80 0x400d0cf0:0x3ffb8e00 0x400810b5:0x3ffb8e20 0x400d0cf0:0x3ffb8ee0 0x400d0cf0:0x3ffb8f60 0x400810b5:0x3ffb8f80 0x400d0cf0:0x3ffb9040 0x400d0cf0:0x3ffb90c0 0x400810b5:0x3ffb90e0 0x400d0cf0:0x3ffb91a0 0x400d0cf0:0x3ffb9220 0x400810b5:0x3ffb9240 0x400d0cf0:0x3ffb9300 0x400d0cf0:0x3ffb9380 0x400810b5:0x3ffb93a0 0x400d0cf0:0x3ffb9460 0x400d0cf0:0x3ffb94e0 0x400810b5:0x3ffb9500 0x400d0cf0:0x3ffb95c0 0x400d0cf0:0x3ffb9640 0x400810b5:0x3ffb9660 0x400d0cf0:0x3ffb9720 0x400d0cf0:0x3ffb97a0 0x400810b5:0x3ffb97c0 0x400d0cf0:0x3ffb9880 0x400d0cf0:0x3ffb9900 0x400810b5:0x3ffb9920 0x400d0cf0:0x3ffb99e0 0x400d0cf0:0x3ffb9a60 0x400810b5:0x3ffb9a80 0x400d0cf0:0x3ffb9b40 0x400d0cf0:0x3ffb9bc0 0x400810b5:0x3ffb9be0 0x400d0cf0:0x3ffb9ca0 0x400d0cf0:0x3ffb9d20 0x400810b5:0x3ffb9d40

Entering gdb stub now.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d0cf0  PS      : 0x00060033  A0      : 0x80085e46  A1      : 0x3ffb8540  
A2      : 0x3ffb85e0  A3      : 0x00000000  A4      : 0x3ffb2334  A5      : 0x383fc000  
A6      : 0x3ff60000  A7      : 0x00000000  A8      : 0x80085800  A9      : 0x0000007d  
A10     : 0x3ffb85e0  A11     : 0x00000000  A12     : 0x3ff40000  A13     : 0x00000000  
A14     : 0x00000030  A15     : 0x3ffb20bc  SAR     : 0x0000001e  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d0cf0:0x3ffb8540 0x40085e46:0x3ffb85c0 0x400810b5:0x3ffb85e0 0x400d0cf0:0x3ffb86a0 0x400d0cf0:0x3ffb8720 0x400810b5:0x3ffb8740 0x400d0cf0:0x3ffb8800 0x400d0cf0:0x3ffb8880 0x400810b5:0x3ffb88a0 0x400d0cf0:0x3ffb8960 0x400d0cf0:0x3ffb89e0 0x400810b5:0x3ffb8a00 0x400d0cf0:0x3ffb8ac0 0x400d0cf0:0x3ffb8b40 0x400810b5:0x3ffb8b60 0x400d0cf0:0x3ffb8c20 0x400d0cf0:0x3ffb8ca0 0x400810b5:0x3ffb8cc0 0x400d0cf0:0x3ffb8d80 0x400d0cf0:0x3ffb8e00 0x400810b5:0x3ffb8e20 0x400d0cf0:0x3ffb8ee0 0x400d0cf0:0x3ffb8f60 0x400810b5:0x3ffb8f80 0x400d0cf0:0x3ffb9040 0x400d0cf0:0x3ffb90c0 0x400810b5:0x3ffb90e0 0x400d0cf0:0x3ffb91a0 0x400d0cf0:0x3ffb9220 0x400810b5:0x3ffb9240 0x400d0cf0:0x3ffb9300 0x400d0cf0:0x3ffb9380 0x400810b5:0x3ffb93a0 0x400d0cf0:0x3ffb9460 0x400d0cf0:0x3ffb94e0 0x400810b5:0x3ffb9500 0x400d0cf0:0x3ffb95c0 0x400d0cf0:0x3ffb9640 0x400810b5:0x3ffb9660 0x400d0cf0:0x3ffb972
.....

```

nkolban, 2017-01-13:
Cancelling due to inability to recreate.

projectgus, 2017-01-13:
I think I can see what would cause this. If the interrupt happens while flash cache is disabled, the panic handler is in IRAM but the gdbstub code is not. So when it jumps to the gdbstub, it triggers an IllegalInstructionException and around we go.

Three possible ways to fix this that I can think of:
* Don't start gdbstub is cache is disabled.
* Re-enable flash cache before starting gdbstub
* Move gdbstub code into IRAM.

Tempted to just do the first one, but interested to know thoughts of @Spritetm and others.

projectgus, 2021-03-03:
This should be fixed now, panic handler re-enables flash cache since 7ff9538c4803f84c876033da9fc60e8a920788a7 (v4.1)

closed 2021-03-03 by projectgus