espressif/esp-idf #233: Crash after update esp-idf and xtensa

michal037, 2017-01-14:
I download the latest esp-idf and xtensa compilator and I compiled example-app:
`git clone https://github.com/espressif/esp-idf-template.git myapp`

After flash:
```
python /home/michal/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/michal/esp32/myapp/build/bootloader/bootloader.bin 0x10000 /home/michal/esp32/myapp/build/app-template.bin 0x8000 /home/michal/esp32/myapp/build/partitions_singleapp.bin
```

ESP has a strange problems:

```
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400fd936
PS      : 0x00060330
A0      : 0x80081a00
............
```

log file: `https://raw.githubusercontent.com/michal037/esp32-for-win-logs/master/esp32_1.log`
I changed nothing in the code. What can cause?

nkolban, 2017-01-14:
If you download a virgin template application and compile/flash it ... does it execute?   There is a thought that the output of "make menuconfig" from previous runs may be incompatible with the current.  Let me suggest the following.

In your application project, save your current "sdkconfig" file and then delete it (the save was in case we need to restore it).  With no current sdkconfig, run "make menuconfig" again and change only those things that may be essential to the running of your app ... it may be none.  At the end, you will have a new "sdkconfig".   Now run "make clean" and perform a complete rebuild and re-deploy.  Does the app still fail?

michal037, 2017-01-14:
Yes, still fail. 
I re-download template-app, remove `sdkconfig` file and I typed `make menuconfig`.
I changed only flash size to `16MB`, and `Debug` to `Release`.
Then I typed `make clean`, `make all` and I have uploaded these flash files to ESP32 memory.

michal037, 2017-01-14:
This happens after the update, earlier flash worked well.

nkolban, 2017-01-14:
Can you run a test with the flash size at the default (2 MB?) ... and without setting "Debug" to "Release"?  Where my mind is going is that, at least for me, I have been working fine ... so I'm fishing for what might be distinct in your environment as compared to mine.  If we can narrow the puzzle to an "aha ... it works with defaults but when we change XYZ, it breaks".

michal037, 2017-01-14:
First what I do, I downloaded from your repository old `esp-idf` and I compiled on this template-app. 
Flash works well! but the problem is It constantly resets connection:
```
I (27127) wifi: reconnect
I (29327) wifi: reconnect
I (31537) wifi: reconnect
I (33737) wifi: reconnect
I (35947) wifi: reconnect
I (38147) wifi: reconnect
I (40357) wifi: reconnect
I (42557) wifi: reconnect
```

As for what you asked me.
I cleaned example app, then remove `sdkconfig` file, I created the file `sdkconfig` with default setting, and then I type `make all flash`.
After uploading the flash, It began to work properly.

michal037, 2017-01-14:
I found the problem, after changing from `Debug` to `Release` begins to occur this error.

When I compile flash with `2MB` size or `16MB` on `Debug`, everything works correctly.
`Release` with different flash size becomes a bug..

I think, is a mistake in the SDK, that causes it.

michal037, 2017-01-14:
Thank you for help. @nkolban

closed 2017-01-17 by igrr