espressif/esp-idf #234: FreeRTOS tick timer selection Timer 2 (int 16, level 5) does not compile

nathanjel, 2017-01-14:
Hi,

If one chooses Timer 2 (int 16, level 5) as the FreeRTOS tick timer in menuconfig, the compilation fails with 
```
In file included from /home/nathan/xnode/esp-idf/components/freertos/include/freertos/xtensa_rtos.h:219:0,
                 from /home/nathan/xnode/esp-idf/components/freertos/./xtensa_init.c:36:
/home/nathan/xnode/esp-idf/components/freertos/include/freertos/xtensa_timer.h:105:4: error: #error "The timer interrupt cannot be high priority (use medium or low)."
   #error "The timer interrupt cannot be high priority (use medium or low)."
    ^
/home/nathan/xnode/esp-idf/make/component_wrapper.mk:176: recipe for target 'xtensa_init.o' failed
make[1]: *** [xtensa_init.o] Error 1
make[1]: *** Waiting for unfinished jobs....
```

The other two options work fine.

Spritetm, 2017-01-15:
Ah, I have to disable that option entirely. This was one of the first options implemented in esp-idf, but at that time we weren't aware that a high-prio interrupt doesn't work as a timer tick source.

nathanjel, 2017-01-15:
If You're keen to let me know which option has to stay as default, I'd be happy to submit a valid pull request with such change.

Spritetm, 2017-01-15:
Ah, thanks for offering, but I already put a merge request in our internal merge queue. Should appear in master somewhere this week.

closed 2017-01-19 by igrr