espressif/esp-idf #235: make monitor not working for esp-idf-v1.0

mazl123321, 2017-01-14:
Hi,

I am trying to use monitor to check the serial port:
```
% make monitor
 make: *** No rule to make target `monitor'.  Stop.
```

Are we dropping the support for the monitor.

FYI:

System information:

uname -a
Darwin xzheng-iMac.local 16.3.0 Darwin Kernel Version 16.3.0: Thu Nov 17 20:23:58 PST 2016; root:xnu-3789.31.2~1/RELEASE_X86_64 x86_64

nkolban, 2017-01-15:
I just installed the latest ESP-IDF and when I run "make monitor" it seems to work just fine for me.    Can you check that you are running "make monitor" in a directory that is generated from an app template.

Spritetm, 2017-01-15:
I think the problem is that mazl123321 is using esp-idf 1.0 instead of the latest. 'make monitor' was only implemented pretty recently, esp-idf 1.0 didn't have it yet. mazl123321: If this is the case, you can get around it by either updating to the latest esp-idf master, or using a random serial terminal program instead.

mazl123321, 2017-01-15:
Yes, thanks guys, I was using the v1.0 release, after updated to master, everything works great.

Thanks!

I am closing out this issue.

closed 2017-01-15 by mazl123321