espressif/esp-idf #238: 's_boot_time_lock' ,  's_boot_time', undeclared, ) newlib\time.c

ESP32DE, 2017-01-15:
hi 

>C:/sdk32/esp-idf/components/newlib/time.c: In function 'set_boot_time':
>C:/sdk32/esp-idf/components/newlib/time.c:94:20: error: 's_boot_time_lock' undeclared (first use in this function)
>     _lock_acquire(&s_boot_time_lock);
>                    ^
>C:/sdk32/esp-idf/components/newlib/time.c:94:20: note: each undeclared identifier is reported only once for each function it appears in
>C:/sdk32/esp-idf/components/newlib/time.c:99:5: error: 's_boot_time' undeclared (first use in this function)
>     s_boot_time = time_us;
>     ^
>C:/sdk32/esp-idf/components/newlib/time.c: In function 'get_boot_time':
>C:/sdk32/esp-idf/components/newlib/time.c:107:20: error: 's_boot_time_lock' undeclared (first use in >this function)
>     _lock_acquire(&s_boot_time_lock);
>                    ^
>C:/sdk32/esp-idf/components/newlib/time.c:111:14: error: 's_boot_time' undeclared (first use in this >function)
>     result = s_boot_time;
>              ^
>C:/sdk32/esp-idf/components/newlib/time.c: At top level:
>C:/sdk32/esp-idf/components/newlib/time.c:92:13: warning: 'set_boot_time' defined but not used >[-Wunused-function]
> static void set_boot_time(uint64_t time_us)
>             ^
>C:/sdk32/esp-idf/components/newlib/time.c:104:17: warning: 'get_boot_time' defined but not used >[-Wunused-function]
> static uint64_t get_boot_time()
>                 ^

best wishes
rudi ;-)

negativekelvin, 2017-01-15:
Is this with menuconfig Timers used for gettimeofday function = none?

ESP32DE, 2017-01-15:
had a look - yes there stands = none
for test i change to FRC1 and i can compile. ( txs for the hint )
it seems if gettimeofday function = none the things are aktive without declaration
best wishes
rudi ;-)

closed 2017-01-17 by igrr