espressif/esp-idf #244: Attribute error 'ImageSegment' when running espcoredump.py

nkolban, 2017-01-15:
I am playing with the new Core dump capabilities and created a dump that was written to UART which I then captured and placed in a text file in base 64 format.  I then ran:
```
python $ESP_IDF_PATH/components/espcoredump/espcoredump.py -p /dev/ttyUSB0 info_corefile -t b64 -c core.dat build/app-template.elf
```

which resulted in 

```
Traceback (most recent call last):
  File "/home/kolban/esp32/esptest/esp-idf/components/espcoredump/espcoredump.py", line 169, in <module>
    class ESPCoreDumpSegment(esptool.ImageSegment):
AttributeError: 'module' object has no attribute 'ImageSegment'
```

This issue asks for the examination of the above command and the resulting output to see where I might have gone wrong and either changes to code if there is a bug or clarification of instructions if needed.

gerekon, 2017-01-16:
Hello Neil,

I am an author of this tool.

Could you check that you have proper IDF_PATH set in your environment? Not ESP_IDF_PATH as you use in command line. espcoredump uses esptool and uses IDF_PATH to locate it.
If this does not help, could you post esptool script you are using with ESP-IDF?

Regards
Aelxey

nkolban, 2017-01-16:
Howdy Aelxey,
I checked that IDF_PATH exists in my environment and it does and it has exactly the same value as ESP_IDF_PATH.  For the record, here is its value:

`/home/kolban/esp32/esptest/esp-idf`

As for which `esptool.py`, I checked and I don't explicitly have any `esptool.py` in my `PATH`.  I am using the very latest version of ESP-IDF as downloaded this morning.  I am assuming that `esptool.py` will be found relative to `IDF_PATH` and if I look in there I see it in `$IDF_PATH/components/esptool_py`.  If  you need a live demonstration, email me (kolban1@kolban.com) or skype (neil.kolban) or find me on `#ESP32` channel on IRC (user: kolban).

gerekon, 2017-01-17:
Hmm... Strange. I sent you contact request via skype.

nkolban, 2017-01-17:
It turned out that I had an old version of esptool installed via "pip".  Running "pip list | grep esptool" showed an old version.  After un-installing with "pip uninstall esptool" ... everything worked exactly as advertized.  I am delighted with the support.  First class!

closed 2017-01-17 by nkolban