espressif/esp-idf #245: GPIO input from pin used for wakeup not readable after wakeup

baycom, 2017-01-15:
There seems to be a problem when a pin is being used for wakeup you cannot read its value after the first wakeup. Until the first deep sleep is being entered the pin can be read flawlessly.

Example:
```
    gpio_set_direction(TEST_PIN, GPIO_MODE_INPUT);
    gpio_set_pull_mode(TEST_PIN, GPIO_PULLUP_ONLY);
    esp_deep_sleep_enable_ext0_wakeup(TEST_PIN, 0);
    printf("GPIO: %d\n", gpio_get_level(TEST_PIN));
    esp_deep_sleep_start();
```
After initial boot you can read:
GPIO0: 1

After first wakeup you can read
GPIO: 0

Is there something I can do or is this a bug?

negativekelvin, 2017-01-16:
Does your code properly config gpio or rely on default config?  Otherwise if in a wake stub reconfig gpio or use rtc_gpio_get_level.

baycom, 2017-01-16:
Thanks, that solved it! Should habe been documented somewhere.

closed 2017-03-15 by igrr