espressif/esp-idf #246: Getting LoadStoreAlignment / IllegalInstruction errors on start then rebooting and fail again..

lpodkalicki, 2017-01-16:
Hi, it happens after successfully flashing any of the ESP-IDF examples (master branch) on board ESP-WROOM-32 compiled with the toolchain xtensa-esp32-elf V5.2.0.

After flash erase and flashing with i.e. 02_blink example a first run goes Ok. - the LED is blinking till next uC resetting / flashing.

```
examples/02_blink$ esptool.py erase_flash
esptool.py v2.0-dev
Connecting...
Detecting chip type... ESP32
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.1s
Hard resetting...
```

```
examples/02_blink$ make flash
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 13376 bytes to 7485...
Wrote 13376 bytes (7485 compressed) at 0x00001000 in 0.7 seconds (effective 161.7 kbit/s)...
Hash of data verified.
Compressed 294848 bytes to 119726...
Wrote 294848 bytes (119726 compressed) at 0x00010000 in 10.5 seconds (effective 224.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 2241.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

However, after a RESTART action triggered by reset button or by cutting off the power supply the module is constant rebooting (see logs for full info)

```
Guru Meditation Error of type IllegalInstruction occurred on core   0. Exception was unhandled.
Register dump:
PC      :  400caade  PS      :  00060730  A0      :  8008a9b1  A1      :  3ffe3b00  
A2      :  400caade  A3      :  3ffb256e  A4      :  00000035  A5      :  00000032  
A6      :  000000ff  A7      :  00000000  A8      :  3f400124  A9      :  0000000c  
A10     :  00000000  A11     :  00000002  A12     :  5fff0007  A13     :  00000000  
A14     :  00000000  A15     :  00000000  SAR     :  00000005  EXCCAUSE:  00000000  
EXCVADDR:  00000000  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  00000000  
Rebooting...
```

And..

```
Guru Meditation Error of type LoadStoreAlignment occurred on core   0. Exception was unhandled.
Register dump:
PC      :  4008aade  PS      :  00060730  A0      :  8008a9b1  A1      :  3ffe3b00  
A2      :  4008aade  A3      :  3ffb256e  A4      :  00000035  A5      :  00000032  
A6      :  000000ff  A7      :  00000000  A8      :  3f400124  A9      :  0000000c  
A10     :  00000000  A11     :  00000002  A12     :  5fff0007  A13     :  00000000  
A14     :  00000000  A15     :  00000000  SAR     :  00000006  EXCCAUSE:  00000009  
EXCVADDR:  4008aade  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  00000000  
Rebooting...
```

Any idea how to fix it ?

negativekelvin, 2017-01-16:
Crashes during phy init are commonly attributable to power supply issues.  What is your setup?

lpodkalicki, 2017-01-16:
My testbed components are:
1. USB to TTL converter based on FTDI FT232 (3.3V, work as a power supply for ESP32), and
2. ESP-WROOM-32 module.

projectgus, 2017-01-16:
FTDI 3.3V regulator output is only rated to 5mA, so it's almost certainly the reason for this crash (the ESP32 and/or the connected flash chip are browning out.) You need a dedicated 3.3V supply that's rated to a couple of hundred mA at least, and ideally some capacitance (100uF or more) on the 3.3V power line.

lpodkalicki, 2017-01-16:
Thanks! I'll back to you with a check result when I test the dedicated 3.3V supply.

lpodkalicki, 2017-01-16:
That was it. Dedicated 3,3V supply fixed this issue. Closing! Thanks for help.

closed 2017-01-16 by lpodkalicki