espressif/esp-idf #250: bluetooth example not working

tianxietaoyan, 2017-01-17:
my phone could see the device when the example is running, but it can not be paired. the output from serial port said:

BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:01:C9:0A
NVDS MAGIC FAILED
RF Init OK with coex
{Esc}[0;31mE (1148) BT: Startup BTU
{Esc}[0m
{Esc}[0;31mE (1148) BT: GATTS_CreateService:  handle of service handle1{Esc}[0m
{Esc}[0;31mE (1158) BT: bta_dm_co_ble_load_local_keys: func not ported
{Esc}[0m
{Esc}[0;32mI (1168) BLUFI_DEMO: BD ADDR: 24:0a:c4:01:c9:0a
{Esc}[0m
{Esc}[0;32mI (1168) BLUFI_DEMO: BLUFI VERSION 0100
{Esc}[0m
{Esc}[0;32mI (1178) BLUFI_DEMO: BLUFI init finish
{Esc}[0m
{Esc}[0;31mE (76928) BT: 
device is connected 08:3d:88:e0:34:8a, server_if=4,reason=0x0,connect_id=4
{Esc}[0m
{Esc}[0;31mE (76928) BT: smp_br_connect_callback is called on unexpected transport 2
{Esc}[0m
{Esc}[0;32mI (76928) BLUFI_DEMO: BLUFI ble connect
{Esc}[0m
{Esc}[0;31mE (106808) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013{Esc}[0m
{Esc}[0;31mE (106808) BT: 
device is disconnected 08:3d:88:e0:34:8a, server_if=4,reason=0x13,connect_id=4
{Esc}[0m
{Esc}[0;31mE (106808) BT: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 083d88e0348a not connected{Esc}[0m
{Esc}[0;31mE (106818) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19
{Esc}[0m
{Esc}[0;31mE (106828) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 08:3d:88:e0:34:8a{Esc}[0m
{Esc}[0;32mI (106838) BLUFI_DEMO: BLUFI ble disconnect
{Esc}[0m
{Esc}[0;31mE (118508) BT: 
device is connected 08:3d:88:e0:34:8a, server_if=4,reason=0x0,connect_id=4
{Esc}[0m
{Esc}[0;31mE (118508) BT: smp_br_connect_callback is called on unexpected transport 2
{Esc}[0m
{Esc}[0;32mI (118508) BLUFI_DEMO: BLUFI ble connect
{Esc}[0m
{Esc}[0;31mE (148348) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013{Esc}[0m
{Esc}[0;31mE (148348) BT: 
device is disconnected 08:3d:88:e0:34:8a, server_if=4,reason=0x13,connect_id=4
{Esc}[0m
{Esc}[0;31mE (148348) BT: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 083d88e0348a not connected{Esc}[0m
{Esc}[0;31mE (148358) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19
{Esc}[0m
{Esc}[0;31mE (148368) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 08:3d:88:e0:34:8a{Esc}[0m
{Esc}[0;32mI (148378) BLUFI_DEMO: BLUFI ble disconnect
{Esc}[0m
{Esc}[0;31mE (1443758) BT: 
device is connected 08:3d:88:e0:34:8a, server_if=4,reason=0x0,connect_id=4
{Esc}[0m
{Esc}[0;31mE (1443758) BT: smp_br_connect_callback is called on unexpected transport 2
{Esc}[0m
{Esc}[0;32mI (1443758) BLUFI_DEMO: BLUFI ble connect
{Esc}[0m
{Esc}[0;31mE (1473398) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013{Esc}[0m
{Esc}[0;31mE (1473398) BT: 
device is disconnected 08:3d:88:e0:34:8a, server_if=4,reason=0x13,connect_id=4
{Esc}[0m
{Esc}[0;31mE (1473398) BT: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 083d88e0348a not connected{Esc}[0m
{Esc}[0;31mE (1473408) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19
{Esc}[0m
{Esc}[0;31mE (1473418) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 08:3d:88:e0:34:8a{Esc}[0m
{Esc}[0;32mI (1473428) BLUFI_DEMO: BLUFI ble disconnect
{Esc}[0m
{Esc}[0;31mE (1491588) BT: btm_ble_resolve_random_addr_on_conn_cmpl unable to match and resolve random address{Esc}[0m
{Esc}[0;31mE (1491838) BT: 
device is connected 6b:38:16:53:7e:ad, server_if=4,reason=0x0,connect_id=4
{Esc}[0m
{Esc}[0;31mE (1491838) BT: smp_br_connect_callback is called on unexpected transport 2
{Esc}[0m
{Esc}[0;32mI (1491838) BLUFI_DEMO: BLUFI ble connect
{Esc}[0m
{Esc}[0;31mE (1525058) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013{Esc}[0m
{Esc}[0;31mE (1525058) BT: 
device is disconnected 6b:38:16:53:7e:ad, server_if=4,reason=0x13,connect_id=4
{Esc}[0m
{Esc}[0;31mE (1525058) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19
{Esc}[0m
{Esc}[0;31mE (1525068) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 6b:38:16:53:7e:ad{Esc}[0m
{Esc}[0;32mI (1525078) BLUFI_DEMO: BLUFI ble disconnect
{Esc}[0m

negativekelvin, 2017-01-17:
What app are you using?  This: http://esp32.com/viewtopic.php?f=13&t=978#p4209 ?

What is your is device and os version?

divinebird, 2017-01-24:
I used example application GATTS Server. There are three devices. Nexus 7 with Android 6.01, Nexus 6 with Android 7, Sony Xperia. I may work with Nexus 7 without any problem. But for tow other devices I see ESP32 but can not connect... I attached bluetooth [dumps](https://github.com/espressif/esp-idf/files/725928/btsnoop_hci_logs.zip) from all devices. You may see clear situation in Nexus 6 dump. After requesting device features it receives disconnection message... For analysing dumps I used wireshark.

divinebird, 2017-01-25:
In bluedroid sources of FreeRTOS I set all debug levels to verbose. While I am connecting to ESP32 from Nexus 6 I see such log:

```
E (47456) BT: BLE HCI(id=62) event = 0x0a)
E (47456) BT: btm_identity_addr_to_random_pseudo
E (47456) BT: btm_ble_resolve_random_addr
E (47456) BT: btm_ble_match_random_bda rec_index = 0
E (47456) BT: sec_flags = 00 device_type = 0
E (47466) BT: btm_ble_match_random_bda rec_index = 1
E (47476) BT: sec_flags = 00 device_type = 0
E (47476) BT: btm_ble_match_random_bda rec_index = 2
E (47486) BT: sec_flags = 00 device_type = 0
E (47486) BT: btm_ble_match_random_bda rec_index = 3
E (47496) BT: sec_flags = 00 device_type = 0
E (47496) BT: btm_ble_match_random_bda rec_index = 4
E (47506) BT: sec_flags = 00 device_type = 0
E (47506) BT: btm_ble_match_random_bda rec_index = 5
E (47516) BT: sec_flags = 00 device_type = 0
E (47516) BT: btm_ble_match_random_bda rec_index = 6
E (47526) BT: sec_flags = 00 device_type = 0
E (47526) BT: btm_ble_match_random_bda rec_index = 7
E (47536) BT: sec_flags = 00 device_type = 0
E (47536) BT: btm_ble_match_random_bda rec_index = 8
E (47546) BT: btm_ble_resolve_address_cmpl p_mgnt_cb->index = 8
E (47556) BT: btm_ble_resolve_random_addr_on_conn_cmpl

E (47556) BT: btm_ble_resolve_random_addr_on_conn_cmpl unable to match and resolve random address
E (47566) BT: btm_ble_connected
E (47576) BT: Security Manager: btm_ble_connected:   handle:0  enc_mode:0  bda:a247176a 
E (47576) BT: btm_sec_alloc_dev

E (47586) BT: BTM_InqDbRead: bd addr [461aa247176a]

E (47586) BT: BTM_GetHCIConnHandle

E (47596) BT: BTM_GetHCIConnHandle

E (47596) BT: l2c_ble_link_adjust_allocation  num_hipri: 0  num_lowpri: 1  low_quota: 10  round_robin_quota: 0  qq: 10
E (47606) BT: l2c_ble_link_adjust_allocation LCB 0   Priority: 0  XmitQuota: 10
E (47616) BT:         SentNotAcked: 0  RRUnacked: 0
E (47626) BT: l2cu_allocate_ccb: cid 0x0000
E (47626) BT: l2c_link_adjust_chnl_allocation
E (47636) BT: POOL ID:3, GKI_poolcount = 400, reserved_buff = 0, weighted_chnls = 2, quota_per_weighted_chnls = 201
E (47646) BT: CID:0x0040 Priority:2 TxDataRate:1 Quota:201
E (47646) BT: btm_find_or_alloc_dev

E (47656) BT: btm_acl_created hci_handle=0 link_role=1  transport=2

E (47666) BT: btm_bda_to_acl found

E (47666) BT: device_type=0x2

E (47676) BT: BLE HCI(id=62) event = 0x04)
E (47676) BT: btm_ble_read_remote_features_complete 
E (55676) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x41d
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00050230  A0      : 0x00000000  A1      : 0x3ffe7b00  
A2      : 0x3ffc31d0  A3      : 0x3ffcce90  A4      : 0x40085b3f  A5      : 0x3ffe7d20  
A6      : 0x00050720  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x03c5a67c  A11     : 0x00249f00  A12     : 0x80084854  A13     : 0x3ffe7d10  
A14     : 0x00000003  A15     : 0x00060723  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffe7b00 0x00000000:0x3ff9c50f

Rebooting...

```

The work breaks on btm_ble_read_remote_features_complete function. And if looks in dump of Nexus 6 BLE working the response for features request come with status: Connection Failed to be Established (0x3e).
[image]

divinebird, 2017-01-26:
I don't know where is the problem. But if you will looking for ESP32 logs

```

E (13522) BT: HCI Enqueue Comamnd opcode=0x2016

E (13522) BT: 16, 20, 02, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00

E (13532) BT: 

E (13542) BT: Transmit Pkt: len 6

E (13542) BT: 01, 16, 20, 02, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00

E (13552) BT: 

E (13552) BT: Recv Pkt: len 7

E (13552) BT: 04, 0f, 04, 00, 05, 16, 20, a5, 00, 00, 00, 00, 10, 00, 00, 80

E (13562) BT: 

E (13562) BT: Recv Pkt: len 15

E (13572) BT: 04, 3e, 0c, 04, 00, 00, 00, 1f, 00, 00, 00, 00, 00, 00, 00, a5

E (13582) BT: 

E (13582) BT: Receive packet event_code=0xf

E (13582) BT: Receive packet event_code=0x3e

E (13592) BT: BLE HCI(id=62) event = 0x04)
E (13592) BT: btm_ble_read_remote_features_complete 
E (13602) BT: HCI Enqueue Comamnd opcode=0x41d

E (13602) BT: 1d, 04, 02, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00

E (13612) BT: 

E (13612) BT: Transmit Pkt: len 6

E (13622) BT: 01, 1d, 04, 02, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00

E (13632) BT: 

E (21632) BT: TimerID 0x3ffc38a8, Name cmd_rsp_to

E (21632) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x41d
```
You will see firstly that Bluetooth library response data without errors "04, 3e, 0c, 04, 00, 00, 00, 1f, 00, 00, 00, 00, 00, 00, 00" for 0x2016. Secondly where is next command HCI Enqueue Comamnd opcode=0x41d that does not come to Android on nexus 6. But it may be because controller of Nexus 6 find error in interconnection and said about in to system with status 0x3e. Necessary deeper analyzing for Bluetooth controllers interconnection...

FayeY, 2017-05-03:
Hi tianxietaoyan & divinebird, is this problem still unsolved?

HITMAnsOFT, 2017-05-07:
@tianxietaoyan, @divinebird, have you tried with any Android 4.4 device? I am currently having issues with Android 6.0.1, while another Android 4.4.4 device connects properly.

Yulong-espressif, 2017-06-03:
@tianxietaoyan 1. Can you update the IDF SDK, I have seen that your log of the crash hci command tout, this issues has been fixed in the new SDK.
2. If you want to use the blufi demo, you should use the specific app(link https://dl.espressif.com/dl/EspBlufi-1.0.apk)to establish connection. Hope this answer can help you.

@HITMAnsOFT I think your log is diffrent issue with @tianxietaoyan, I find that the disconnect reason code = 0x08(Connection time out error), and before the disconnet, gatt_disc_cmpl_cback() - Register for service changed indication failure has been print. So I think this is because of the client sent req to the server and it is dont sent back the rsp. We will fixed this issues in the next version of SDK.

HITMAnsOFT, 2017-06-07:
@Yulong-espressif Thanks for your response! Should I file a new issue? Do you need any extra information?

Yulong-espressif, 2017-06-11:
@HITMAnsOFT Of course, what is your new issue? If you have the log, you can show it to me, that will be more helpful.

HITMAnsOFT, 2017-06-11:
@Yulong-espressif No, I meant the problem described in my previous comment on May 7. Should I move the log into a new issue?