espressif/esp-idf #251: compiling problems because of kconfig on mac os sierra

eliabieri, 2017-01-17:
When I want to compile using ```make menuconfig``` , the compiler throws the following error:
```In file included from zconf.tab.c:237:
./zconf.hash.c:166:1: error: conflicting types for 'kconf_id_lookup'
kconf_id_lookup (register const char *str, register size_t len)
^
./zconf.hash.c:33:31: note: previous declaration is here
static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
                              ^
1 error generated.
make[1]: *** [zconf.tab.o] Error 1
```

igrr, 2017-01-17:
Could you please check the suggestions mentioned here and see if they fix the issue?
https://github.com/pfalcon/esp-open-sdk/issues/236#issuecomment-271697219

jtlobdell, 2017-02-15:
I had this same issue. I modified the declaration in zconf.hash.c to use size_t instead of unsigned int, and that got it to build.

projectgus, 2017-02-15:
gperf 3.1 changed the type signature for this function, for 64-bit compatibility reasons.

A permanent fix is pending and will be in master soon.

In the meantime you can delete this entire line from zconf.gperf and it will fix the problem:
https://github.com/espressif/esp-idf/blob/master/tools/kconfig/zconf.gperf#L12

(Editing zconf.hash.c will work until you `make clean`, because this file is generated from zconf.gperf.)

closed 2017-02-15 by igrr

oaky87, 2017-03-26:
same problem