espressif/esp-idf #253: First boot failure after encryption

rojer, 2017-01-17:
I enabled flash encryption but not secure boot.
I burned a reusable key (so i can reflash in development), flashed plain text FW and booted the device, which proceeded to encrypt app partitions but not OTA data.
Subsequent boot failed due to, i think, `otadata` partition not having been encrypted.
This is not mentioned in the guide and i think it should just be encrypted unconditionally, just like app partitions, since it is accessed by the boot loader using mmap.

```
I (47) boot: ESP-IDF v1.0-374-gf9eff70 2nd stage bootloader
I (47) boot: compile time 15:55:31
I (47) boot: Enabling RNG early entropy source...
I (67) boot: SPI Speed      : 40MHz
I (79) boot: SPI Mode       : DIO
I (92) boot: SPI Flash Size : 4MB
I (104) boot: Partition Table:
I (115) boot: ## Label            Usage          Type ST Offset   Length
I (138) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (161) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (185) boot:  2 app_0            OTA app          00 10 00010000 00100000
I (208) boot:  3 fs_0             Unknown data     01 82 00110000 00020000
I (231) boot:  4 app_1            OTA app          00 11 00130000 00100000
I (254) boot:  5 fs_1             Unknown data     01 82 00230000 00020000
I (278) boot: End of partition table
I (397) boot: Checking flash encryption...
W (397) flash_encrypt: Using pre-loaded flash encryption key in EFUSE block 1
I (401) flash_encrypt: Setting CRYPT_CONFIG efuse to 0xF
I (455) flash_encrypt: Disable UART bootloader encryption...
I (456) flash_encrypt: Disable UART bootloader decryption...
I (465) flash_encrypt: Disable UART bootloader MMU cache...
I (484) flash_encrypt: Disable JTAG...
I (2378) flash_encrypt: Encrypting partition 2 at offset 0x10000...
I (40813) flash_encrypt: Encrypting partition 4 at offset 0x130000...
ts Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:5048
load:0x40078000,len:9020
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
I (92) boot: ESP-IDF v1.0-374-gf9eff70 2nd stage bootloader
I (92) boot: compile time 15:55:31
I (92) boot: Enabling RNG early entropy source...
I (111) boot: SPI Speed      : 40MHz
I (124) boot: SPI Mode       : DIO
I (137) boot: SPI Flash Size : 4MB
I (150) boot: Partition Table:
I (161) boot: ## Label            Usage          Type ST Offset   Length
I (184) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (207) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (230) boot:  2 app_0            OTA app          00 10 00010000 00100000
I (253) boot:  3 fs_0             Unknown data     01 82 00110000 00020000
I (277) boot:  4 app_1            OTA app          00 11 00130000 00100000
I (300) boot:  5 fs_1             Unknown data     01 82 00230000 00020000
I (323) boot: End of partition table
E (336) boot: ota data partition info error
user code done
```

rojer, 2017-01-17:
so, i fixed it by manually encrypting 8192 bytes of 0xff and writing the result at 0xd000
however, [this code](https://github.com/espressif/esp-idf/blob/master/components/bootloader/src/main/bootloader_start.c#L314) writes default ota data in the clear, not encrypted, so the subsequent boot failed too.
final solution to get encrypted boot working was to encrypt and write non-empty ota data.

projectgus, 2017-01-17:
Thanks @rojer. We'll fix this so OTA & encryption can work together.

FayeY, 2017-07-10:
Is this problem solved now?

projectgus, 2017-07-20:
I checked this in IDF e468cdee1d8b15f (which reworks a lot of the boot process to optimise performance) and was able to run OTA updates with flash encryption on. The commit should be on github master shortly, and will be in the IDF V3.0 release.

Will comment separately on the other OTA data changes.

closed 2017-07-20 by projectgus