espressif/esp-idf #259: Suggestion: Alternate names for GPIO pull modes

nkolban, 2017-01-17:
Currently, the possible values for the pull mode of a GPIO pin are defined as:

* GPIO_PULLUP_ONLY
* GPIO_PULLDOWN_ONLY
* GPIO_PULLUP_PULLDOWN
* GPIO_FLOATING

This issue asks for consideration of a name change for the first two.  The suggested new names are:

* GPIO_PULLUP
* GPIO_PULLDOWN

michal037, 2017-01-17:
I think there must be third state, "High Z". where voltage is indefinite, between 3V3 to 0V.

michal037, 2017-01-17:
i gave a bad example, I thinking about 4 state, where 3 state doesn't have be.

negativekelvin, 2017-01-18:
Changing the name without changing the behavior would create a source of possible confusion.  If one did:

```
gpio_set_pull_mode(#, GPIO_PULLUP)
gpio_set_pull_mode(#, GPIO_PULLDOWN)
```

One may expect it equal to :

`gpio_set_pull_mode(#, GPIO_PULLUP_PULLDOWN)`

Whereas 

```
gpio_set_pull_mode(#, GPIO_PULLUP)
gpio_set_pull_mode(#, GPIO_PULLDOWN_ONLY)
```

Is obvious that the pullup is disabled

nkolban, 2017-01-18:
Thank you NegativeKelvin, that does make sense.  Let's leave this issue open for another week or so and see if there are any additional opinions or thoughts, but if not, I'll just close it out.    It hadn't occurred to me that gpio_set_pull_mode() might be thought of as "additive" by consumers ... but if that is a consideration, then the names including the phrase "ONLY" do indeed make sense.

FayeY, 2017-05-03:
Hi nkolban, can we close this issue now?

nkolban, 2017-05-03:
okly dokly ... thanks for the reminder.

closed 2017-05-03 by nkolban