espressif/esp-idf #261: Startup issue in NVS when compiled with compilation optimizations on

nathanjel, 2017-01-18:
After enabling optimizations to the code, this is the result

Serial port
```
Register dump:�Ѐ����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
PC      : 0x400f419e  
PS      : 0x00060330  
A0      : 0x80081e48  
A1      : 0x3ffe3af0  
A2      : 0x3ffbb17c  
A3      : 0x4008198c  
A4      : 0x00000000  
A5      : 0x3f404000  
A6      : 0x3f400bbc  
A7      : 0x00000001  
A8      : 0x800f419c  
A9      : 0x3ffe3ad0  
A10     : 0x00000164  
A11     : 0x00000164  
A12     : 0x00000000  
A13     : 0x00000000  
A14     : 0x3ffb5e6c  
A15     : 0x00024e94  
SAR     : 0x00000000  
EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  
LBEG    : 0x40001609  
LEND    : 0x4000160d  
LCOUNT  : 0x00000000  

Backtrace: 0x400f419e:0x3ffe3af0 0x40081e48:0x3ffe3b10 0x400f3a38:0x3ffe3b60 0x400f3d89:0x3ffe3bc0 0x400f2ba0:0x3ffe3c20 0x400f2a7c:0x3ffe3c70 0x400f2ac2:0x3ffe3c90 0x40080b70:0x3ffe3cc0 0x40080d2d:00

Entering gdb stub now.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d1bc2  
PS      : 0x00060033  
A0      : 0x80086470  
A1      : 0x3ffe3990  
A2      : 0x3ffe3a30  
```

No chance to run GDB, so addr2line
```
nathan@ldev:~/xnode/SNESP32$ addr2line -e ./build/stripnode.elf 
0x400f419e:0x3ffe3af0
flash_ops.c:?
0x40081e48:0x3ffe3b10
/home/nathan/xnode/esp-idf/components/spi_flash/./flash_ops.c:303
0x400f3a38:0x3ffe3b60
/home/nathan/xnode/esp-idf/components/nvs_flash/src/nvs_page.cpp:812
0x400f3d89:0x3ffe3bc0
/home/nathan/xnode/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp:28
0x400f2ba0:0x3ffe3c20
/home/nathan/xnode/esp-idf/components/nvs_flash/src/nvs_storage.cpp:241
0x400f2a7c:0x3ffe3c70
/home/nathan/xnode/esp-idf/components/nvs_flash/src/nvs_api.cpp:70
0x400f2ac2:0x3ffe3c90
/home/nathan/xnode/esp-idf/components/nvs_flash/src/nvs_api.cpp:87
0x40080b70:0x3ffe3cc0
/home/nathan/xnode/esp-idf/components/esp32/./cpu_start.c:273
0x40080d2d
/home/nathan/xnode/esp-idf/components/esp32/./cpu_start.c:138

```

igrr, 2017-01-18:
Do you have this new commit in your tree: ce8ec33? The issue looks very much like https://github.com/espressif/esp-idf/issues/233.

nathanjel, 2017-01-18:
Roger that. My bad. Thanks!

closed 2017-01-18 by nathanjel