espressif/esp-idf #262: blufi example not compiling

seopyoon, 2017-01-18:
I have updated the esp-idf and toolchain.  

But, when I try to `make flash`, it gives me following error: `blufi/main/./blufi_main.c:28:27: fatal error:esp_blufi_api.h: No such file or directory`  

Perhaps I have done something wrong in setting up the environment, or something is wrong with the make file?  I use Visual Studio Code for developing, and the IDE finds these header files with no problem.  Somehow the makefile cannot.

negativekelvin, 2017-01-18:
I get this error when Bluetooth component is not enabled in menuconfig

seopyoon, 2017-01-18:
I am away from work, so I cannot check it for a couple of days, but from `sdkconfig.defaults`, the Bluetooth component should be already enabled, unless I explicitly disabled it during `make menuconfig` which I did not.  @negativekelvin does your blufi example work with the new sdk??

negativekelvin, 2017-01-18:
Yes it successfully builds for me with the Bluetooth component enabled

seopyoon, 2017-01-20:
I had to somehow delete the whole esp-idf folder and re clone it.  pulling and loading submodules weren't enough.  Thanks for input anyways.  Cheers.

closed 2017-01-20 by seopyoon