espressif/esp-idf #265: ESP32-WROVER SPI Driver

Byteschmiede, 2017-01-19:
Hi devs,

Is there any ongoing development of an officle SPI Driver for the LCD Display? Espressif is selling the DevKitJ aka ESP32-WROVER-Kit without any offical drivers or api for the onboard display.

I already used the "spi_master" example by rudi but this is very very complex and not even fast enough to do "good" stuff on it. Even to draw text or something is very hard because you have to render it and provide the RGB values of every pixel, line by line.

It would be nice if someone could integrate the Adafruit solution of the ili9341 display.

Have a nice day!

projectgus, 2017-01-26:
Hi @Byteschmiede,

A high level API for the ILI9341 is coming, and will be supported in ESP-IDF.

There is low-level ILI9341 support in the [NESEMU code](https://github.com/espressif/esp32-nesemu/), and using Arduino libraries (either via the arduino-esp32 esp-idf component or the Arduino IDE) also works but is slow as you note.

We'll update this issue once an integrated option is available.

Byteschmiede, 2017-01-26:
Hi @projectgus 
Thank you !

sfranchin63, 2017-04-14:
Hi @projectgus,
you wrote that high-level API for the ILI9341 is coming. Can you tell me when the API will be ready.
Thank yuo

sukeshak, 2017-08-29:
@Byteschmiede Have you tried this one? Since it says ILI9341 is supported.
https://github.com/loboris/ESP32_TFT_library

By next week I will be testing it... Waiting for my ESP32 to arrive :)

krzychb, 2018-04-04:
I see ILI9341 LCD display support implemented under https://github.com/espressif/esp-iot-solution/tree/master/components/spi_devices/lcd

closed 2018-09-17 by Alvin1Zhang