espressif/esp-idf #266: deep sleep wakeup documentation inconsistency

BurningEnlightenment, 2017-01-19:
Within the [External wakeup (ext1) docs](https://esp-idf.readthedocs.io/en/latest/api/system/deep_sleep.html#external-wakeup-ext1) @igrr wrote the following:
> * wake up if any of the selected pins is low
> * wake up if all the selected pins are high

But the `esp_deep_sleep_enable_ext1_wakeup()` and `esp_ext1_wakeup_mode_t` docs state something contrary:
> ESP_EXT1_WAKEUP_ALL_LOW: wake up when all selected GPIOs are low
> ESP_EXT1_WAKEUP_ANY_HIGH: wake up when any of the selected GPIOs is high

igrr, 2017-01-19:
Thanks for pointing this out. Enum definitions are correct in this case.
I will update the docs.

krzychb, 2017-05-25:
@FayeY, I do not see this being updated in docs v2.0

[image]

krzychb, 2017-06-19:
@FayeY, this issue is fixed in 2.1 release and  the latest documentation, I would close it.

[image]

closed 2017-06-20 by FayeY