espressif/esp-idf #270: crosstool-ng out of date for repository

RobinCutshaw, 2017-01-20:
I just did a git pull on the repository and it gives a warning when trying to build examples:

robin@u64imac:~/esp/esp-idf/examples/protocols/http_request$ make
WARNING: Toolchain version is not supported: crosstool-NG 8d95cad
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0

Version 1.22.0-59 is still the current download and the one documented as the current download.
Thread model: single
gcc version 4.8.5 (crosstool-NG 8d95cad)

negativekelvin, 2017-01-20:
Docs seem to show new toolchain: http://esp-idf.readthedocs.io/en/latest/index.html

projectgus, 2017-01-20:
Hi Robin,

We've updated the documentation in the repository (which @negativekelvin has linked to.) Unfortunately the ESP-IDF Getting Started PDF hasn't yet been updated (we are working on a solution there to document multiple versions, as v1.0 still expects gcc 4.8.5 and the "master" branch is prerelease.) Is the PDF the docs you are referring to, or have we missed somewhere?

Angus

RobinCutshaw, 2017-01-20:
Yes, it is: http://www.espressif.com/sites/default/files/documentation/esp-idf_getting_started_guide_en.pdf from the document page: http://www.espressif.com/en/support/download/documents?keys=&field_type_tid%5B%5D=13

Robin

closed 2017-11-21 by FayeY