espressif/esp-idf #271: Problem including regex.h from newlib

ivanO86, 2017-01-20:
Hi,

I am trying to validate some regular expressions and I found the regex library into the newlib include folder. When I try to include this header in my code, I have some unknown type errors:

~/esp-idf/components/newlib/include/regex.h:43:1: error: unknown type name 'off_t'
 typedef off_t regoff_t;

Also I have tried to find the regex symbols into the newlib build directory but I couldn't find anything.

Can we use this library? Am I missing any link flags or something like that? I use the time.h and works properly.

Regards.

negativekelvin, 2017-01-20:
#include <sys/types.h>

ivanO86, 2017-01-20:
Hi,

thanks for the comment. I have added this header, but I still having problems with the linker. 
undefined reference to `regcomp'
undefined reference to `regexec'
undefined reference to `regerror'
undefined reference to `regfree'

I am using this library in my own component and I don't know if I am missing some linker flag, but I cannot find these symbols in the newlib build folder.

regards.

negativekelvin, 2017-01-20:
Ah OK sorry don't know if the library was built with that option

closed 2017-07-07 by FayeY

HrMitrev, 2019-05-14:
I need regex, please, much please add this functionality