espressif/esp-idf #273: VFS and LWIP should share file/socket descriptor space

rojer, 2017-01-21:
Currently some functions of the POSIX file API are routed to VFS and those that are partof the BSD socket API are routed to VFS. this is very confusing. for example, it is possible to `open` a file, but subsequent `read` will fail because it goes to LWIP.
of course, one can side-step the issue by using buffered file i/o (`fopen`, `fread`), but it's not ideal. it should be possible to route LWIP through VFS and make them share descriptor space.

young-mu, 2017-02-28:
Hi, @rojer. I think you can consider the LWIP build option **LWIP_SOCKET_OFFSET** to let socket descriptor starting from an offset to avoid the descriptor overlap with VFS file descriptor.

rojer, 2017-02-28:
the issue is not overlap of descriptor space, but the fact that `read` is routed to LWIP only and cannot be used for operations on VFS files.

young-mu, 2017-02-28:
Okay.

jsarrett, 2017-04-20:
I also have run in to this problem.  It's very serious because it means basically that libc is linking unreliably.  It's not at all clear which read or write will be linked to a component.  LWIP_SOCKET_OFFSET should make it easy enough to fix inside of newlib, but that call routing *has* to be done at runtime.

closed 2017-10-18 by igrr

chanibal, 2017-11-13:
@igrr, in a [forum post from half a year ago](https://www.esp32.com/viewtopic.php?t=2317), you mentioned that you'll fix the LWIP macro issues with this ticket. As far as I see in the closing commit, that was not the case.

I know, there is the `#undef` hack, but frankly it's very annoying to have issues with such common indentifiers. Not only `bind`, but also the lack of `send` is quite annoying.

Any chance to fix this, perhaps attach it to another ticket?

projectgus, 2017-12-14:
@chanibal
> you mentioned that you'll fix the LWIP macro issues with this ticket. As far as I see in the closing commit, that was not the case.

Sorry we missed this comment on this closed issue until just now. I agree we should fix this properly. Could you please open a new issue to keep track of this one?