espressif/esp-idf #278: SSL_read overshooting?

ksitko, 2017-01-21:
I'm trying to read the header of a .bin file hosted in a S3 bucket for the purpose of confirming the MD5 for OTA. After sending the HEAD HTTP request I use SSL_read to read the result, however I'm getting a bunch of junk at the end of the string.

This is the result of a single SSL_read command:

`x-amz-id-2: 9/EqFVJr9gIJ/0BYIT0nhOpHVEyd6zRwXIAWxCnBwj5N02j3dgRo+JgYui5TmiC0anOWm/rBID8=
x-amz-request-id: 957A1A938138DCB3
Date: Sat, 21 Jan 2017 21:50:13 GMT
Last-Modified: Wed, 18 Jan 2017 23:39:41 GMT
ETag: "e43ccaad46f7be9e08b1a7ac24855921"
Accept-Ranges: bytes
Content-Type: application/octet-stream
Content-Length: 294416
Server: AmazonS3

▒▒1h▒
▒▒▒▒▒▒▒?▒0▒?`

Before I had a loop which would rely on the output of SSL_read being positive but that completely overshoot and started outputing an enormous amount of junk.

Is this a problem with SSL_read? Or should SSL_read not be used for a request like this?

negativekelvin, 2017-01-22:
SSL_read should return the number of bytes it successfully read so you should know where the valid data ends.  Otherwise if you calloc the read buffer you should have a null termination.

ksitko, 2017-01-22:
If I leave it with a large enough buffer it appears to read the entirety of the .bin file I'm trying to pull header data for, which makes no sense as the server should only return the header as a response to a HEAD request. Having a buffer size of 400 to 800 will return only the header. But if have a buffer over 800 then it starts pulling what I assume is the bin data.

negativekelvin, 2017-01-22:
Well supposedly s3 supports head http://docs.aws.amazon.com/AmazonS3/latest/API/RESTObjectHEAD.html

But to see if it is a server side  issue try your code on something else like Google.com which seems to work with head requests

yschaeff, 2017-01-23:
I think negativekelvin points you to the right thing. It is not about the size of the buffer. You are interpreting the data in the buffer as a string. But it is not a string. Terminate it with a zero to make it into a string: buf[<ssl_read return value>] = 0;

ksitko, 2017-01-23:
Yup, I initialized all my buffers with zero, save this one. Opps.

closed 2017-01-23 by ksitko