espressif/esp-idf #279: xtensa-esp32-elf-linux32-1.22.0-61-gab8375a-5.2.0.tar.gz not found on espressif dl server

bvernoux, 2017-01-22:
I'm using latest information to upgrade the toolchain on Ubuntu 32bits
http://esp-idf.readthedocs.io/en/latest/linux-setup.html#step-1-download-binary-toolchain-for-the-esp32
for 32-bit Linux:
https://dl.espressif.com/dl/xtensa-esp32-elf-linux32-1.22.0-61-gab8375a-5.2.0.tar.gz
Problem is this link returns "404 Not Found"

projectgus, 2017-02-03:
The 32-bit toolchain download link has been fixed. Sorry for the inconvenience.

closed 2017-02-03 by projectgus

btukacs, 2020-08-12:
Hi, i am following setup process for 3.3: https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/macos-setup.html
https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-96-g2852398-5.2.0.tar.gz
gives me also 404. Please repair it asap.

Alvin1Zhang, 2020-08-13:
@btukacs Thanks for reporting, we will check.