espressif/esp-idf #280: How to create multiple characteristic on same service(for Bluetooth  GATT server)

YanMinge, 2017-01-22:
I try to add characteristic with "esp_ble_gatts_add_char". but only the first characteristic can be searched.

negativekelvin, 2017-01-22:
Have you looked to gatt_server_service_table example?

divinebird, 2017-01-23:
When you use esp_ble_gatts_create_service there is last argument where you set handles count for this service. In example it is 4. I found that for every characteristic use 2 handles as service and descriptor use 1 handler. So if you like to add characteristic you should init service with 6 or more handle number.

YanMinge, 2017-01-23:
Thanks，it can work now

closed 2017-01-23 by YanMinge

sowmimalu, 2019-11-27:
Can you please guide me on gatt  client ??

chegewara, 2019-11-28:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_client/tutorial/Gatt_Client_Example_Walkthrough.md#getting-characteristics