espressif/esp-idf #281: Windows toolchain question

MartyMacGyver, 2017-01-23:
I notice that the Arduino ESP32 project references https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip for Windows now. Is that a late bugfix revision of the Windows toolchain, a fork specifically for that project, or something else?

If it's a bugfix for Windows that is more generally applicable to the ESP32 toolchain, the issue is that the esp32_win32_msys2_environment_and_toolchain distribution should be updated with it, as well as the links on http://esp-idf.readthedocs.io/en/latest/windows-setup.html

negativekelvin, 2017-01-23:
No they are the same file, have same hash.

igrr, 2017-01-23:
Thanks for mentioning the all-in-one zip file, it needs to be updated.

MartyMacGyver, 2017-01-23:
That's a retroactive update then... I have xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip from a few days ago (build date 2017-01-11) and it definitely is different in terms of contents and build dates internally.

Likewise, I have esp32_win32_msys2_environment_and_toolchain-20170111.zip from the same time, and the toolchain files within it were the same 2017-01-11 ones (which appears to be what you mean).

I'll ping the arduino-esp32 project about this, as having multiple toolchains (even if they differ only by name) is apt to confuse people, particularly those who grabbed xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip before it was revised (not everyone uses the given all-in-one zip).

igrr, 2017-01-23:
Yes, I'll update the links, no need to open an issue in the Arduino repo. Thanks.

negativekelvin, 2017-01-23:
So if the binaries have changed is it a rebuild with no changes, only new timestamps?

igrr, 2017-01-23:
The original toolchain published as xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip wasn't functional as it was dynamically linked against a non-existent pthreads library. When this was reported by users in the Arduino project, I have rebuilt the toolchain with the correct flags, and this was uploaded as xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip. Because I was using a script to build and upload things to the download server, i think i may have overwritten the original xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip before actually uploading xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip...

At this point it doesn't make sense to restore the original xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip as the toolchain in that archive would not even run. I will update the download links to point to the -2.zip version though, to avoid further confusion, and will update the all-in-one msys environment.

MartyMacGyver, 2017-01-23:
I wasn't suggesting restoring the broken one... I simply didn't know what -2 was and having the original one I wondered which was which.

But that means the one in the esp32_win32_msys2_environment_and_toolchain-20170111.zip is broken too... yet it's been working fine for me, or appeared to be. Perhaps because I'm not using affected functionality?

And given that, would this have broken tasks (e.g., xTaskCreate)?

igrr, 2017-01-23:
I was referring to this issue: https://github.com/espressif/arduino-esp32/issues/135#issue-201093487

Basically the compiler didn't run at all. No output files were produced, so 'affected functionality' is more or less equal to "compiling any ESP32 program". I suppose it can still work for you if you happen to have the missing dynamic library in your environment, installed in some other way, which other users didn't have.

negativekelvin, 2017-01-23:
For reference I had just dropped it into previous msys2 aio.

MartyMacGyver, 2017-01-23:
Weird... maybe I do somewhere, but the MSYS2 is pretty well locked down path-wise. Either way, thanks for clarifying this!

igrr, 2017-01-23:
Well, MSYS *is* your environment, right? The issue was reported by an Arduino user. Arduino IDE doesn't run the compiler from MSYS environment, it launches it straight from Java.

negativekelvin, 2017-01-23:
This has been a fun ghost chasing adventure

MartyMacGyver, 2017-01-23:
@igrr - I'm aware of that. I asked why the toolchain was changed in this way because I noticed it in Arduino first, then noticed MSYS2 contains the release of the toolchain that basically disappeared because it's broken (as you noted above). I wanted to understand what happened there, get the Arduino docs straightened out in that regard, and get the MSYS2 distribution updated with the new, not-broken compiler (as also acknowledged above).

It's possible that I never happened to run into the buggy portion of the compiler in MSYS2, but others may well (it's unlikely my system was finding the missing parts anywhere else either.)

igrr, 2017-01-23:
Well, the point i guess is that when the old build wasn't working in Arduino due to the missing library but was working in MSYS2 because it happens to have that same library. This is why the issue happens in Arduino but not in MSYS2.

MartyMacGyver, 2017-01-23:
Ah. I see that both releases of `xtensa-esp32-elf` have exactly the same files, though a handful were built differently (as per the change in config). I thought perhaps that config change might have some unwanted effect in MSYS2, as part of this fix was not just files outside the `xtensa-esp32-elf` but the configuration of the toolchain itself - so I asked about that in addition to my original question relative to Arduino-ESP32.

closed 2017-03-28 by igrr

jjimmy-S, 2018-03-25:
Hi @igrr I was wondering if there is a new alternative to solve this error, I have tried with the default toolchain, and the old version that you suggest, but I still get the same error, "Error compiling for board ESP32 Dev Module."  can you help me?

commit f5aee6a6 (2017-03-28): windows config: Update Windows setup script, pre-installed image, instructions.

Add notes about MSYS2 package mirrors & HTTP proxies for users in China.

Closes #260 https://github.com/espressif/esp-idf/issues/260
Closes #281 https://github.com/espressif/esp-idf/issues/281
Closes #200 https://github.com/espressif/esp-idf/issues/200
Closes #430 https://github.com/espressif/esp-idf/pull/430
Accomplishes similar goals to #264 https://github.com/espressif/esp-idf/pull/264