espressif/esp-idf #282: Possible memory leak? Or lack of understanding.

seopyoon, 2017-01-23:
Hello, I am trying out the blufi example with the apk provided in this [thread](http://esp32.com/viewtopic.php?f=13&t=978&p=4447#p4447).

I can successfully connect to AP and get IP.  But, when I add a simple task, as below,

```
static void mainLoop(void *pvParameters)
{
    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
    BLUFI_INFO("Connected to AP.  Main Loop Begins...\n");

    while(1) {
        BLUFI_INFO(".");
        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
}
``` 
as a task at the very end of the line as follows:
`xTaskCreate(&mainLoop, "MAIN LOOP", 512, NULL, 5, NULL);`

After a while, perhaps less than a minute after the task begins, the processor restarts, throwing an error saying that stack overflow was detected.
```

***ERROR*** A stack overflow in task MAIN LOOP has been detected.
abort() was called at PC 0x4008652c
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40158172:0x3fff4f40 0x40084a6c:0x3fff4f60 0x40084694:0x3fff4f80 0x40085d2f:0x3fff4fa0 0x40085dc0:0x3fff4fc0 0x40104c3b:0x3fff4fe0

Rebooting...
```

Why would this happen?? All I do is print a character every half a second.

seopyoon, 2017-01-23:
To add, if I add a GPIO-modifying code instead of the print, it didn't crash for half an hour.  (Such as LED control)

igrr, 2017-01-23:
As the message indicates, you are getting a stack overflow. 512 bytes of stack is way too small for a `printf`, try increasing stack size to, say, 2048 bytes.

seopyoon, 2017-01-23:
@igrr thank you for your reply.  Does that mean that the stack keeps growing as I print something?? So, if I were to test, and have it print something for a long period of time, the stack size should be infinitely large?

negativekelvin, 2017-01-23:
No stack grows and shrinks so once you find proper stack size it should not overflow.

seopyoon, 2017-01-24:
Got it!

closed 2017-01-24 by seopyoon