espressif/esp-idf #283: SPI 0 is not working in esp32

amallal, 2017-01-23:
Hi
i tried to work with all spi in esp32, I wrote a code to initialize spi and send one data , it is working with both HSPI and VSPI in esp32, but same i tried in normal SPI (Pins 6,7,8,9,10,11), but it is not working. Could you please help to fix this issue
I am attaching my code here in Arduino. please find the attachment.
[spi_test.tar.gz](https://github.com/espressif/esp-idf/files/722901/spi_test.tar.gz)

nkolban, 2017-01-23:
Howdy, if we look at the Overview text found here .. .http://esp-idf.readthedocs.io/en/latest/api/peripherals/spi_master.html

We find that there are four SPIs known as SPI0, SPI1, HSPI and VSPI.  In the overview there is a restriction that states that *only* HSPI and VSPI may be used for application purposes.

projectgus, 2017-03-07:
Hi @amallal ,

Sorry it took so long for someone for the IDF team to get back to you here.

* The normal SPI pins (SPI0, the pins you mentioned) are used for connection to the internal SPI flash where the program code is located (and mapped via flash cache). It is possible to use this SPI bus to connect to other devices, but it's complex (as code also runs from there) and is not supported in either Arduino or IDF.
* For issues with Arduino, in future please open tickets against the Arduino project here: https://github.com/espressif/arduino-esp32/issues/

For now, @nkolban's advice to use HSPI/VSPI only (with multiple CS lines on the same SPI bus if you need to connect multiple devices) is the best advice to follow.

Thanks,

Angus

closed 2017-03-07 by projectgus

rmetzner49, 2017-06-02:
I have a portion of my project working, I can display graphics on an East Rising 5" SPI Display using VSPI.  When I attempt to connect my DS3234 to the same basic pins EXCEPT a different SS, I'm having interference.  Should I be running both instances of the SPI without attaching the SS pin and then just banging it on and off when needed?  Should I be creating different instances with attach/detach commands?