espressif/esp-idf #284: Working with descriptor of characteristic not full

divinebird, 2017-01-23:
When you like to create descriptor for characteristics there is not one link to this characteristic. It is significantly when you try to create many characteristics on one service. When you create descriptors at the finish of creating all characteristics they all(descriptors) will attached to last created characteristic. When ESP_GATTS_ADD_CHAR_DESCR_EVT event come also there is no information about characteristic that was been attached descriptor.

divinebird, 2017-01-23:
Looking for the sources I may come to think that there is no chance to add such logic for a while. But it is very bad... I found only one resolution - add characteristics and connected descriptors one by one.

TianHao-Espressif, 2017-02-06:
If you use "esp_ble_gatts_add_char" or "esp_ble_gatts_add_char_descr", you should call them one by one. It is bluedroid original way. Then we have done the other measure to add the attribute by attribute-table which released on ESP-IDF v2.0. You can try it and see the demo in examples/bluetooth/gatt_server_service_table.

hmj844263150, 2017-04-24:
I have learn BLE at esp32 nowadays, and got same doubt with you @divinebird, I want to add a new characteristic in a service with gatt_server. 
It seems have to create characteristics one by one, and you can get some information with the function:esp_ble_gatts_get_attr_value() by handle you can get in event param. and i have try to coding it:
`    
case ESP_GATTS_CREATE_EVT:
        ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT, status %d,  service_handle %d\n", param->create.status, param->create.service_handle);
        gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.len = ESP_UUID_LEN_16;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A;

        esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);

        esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid,
                               ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                               ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY, 
                               &gatts_demo_char1_val, NULL);

		
        break;
    case ESP_GATTS_ADD_INCL_SRVC_EVT:
        break;
    case ESP_GATTS_ADD_CHAR_EVT: {
		uint16_t length = 0;
        const uint8_t *prf_char;
		
		ESP_LOGI(GATTS_TAG, "ADD_CHAR_EVT, status %d,  attr_handle %d, service_handle %d\n",
		                param->add_char.status, param->add_char.attr_handle, param->add_char.service_handle);
        esp_ble_gatts_get_attr_value(param->add_char.attr_handle,  &length, &prf_char);
        ESP_LOGI(GATTS_TAG, "the gatts demo char length = %x\n", length);
        for(int i = 0; i < length; i++){
            ESP_LOGI(GATTS_TAG, "prf_char[%x] =%x\n",i,prf_char[i]);
        }
		
		if(param->add_char.attr_handle == param->add_char.service_handle+2){
			esp_bt_uuid_t descr_uuid;
			descr_uuid.uuid.uuid16 = ESP_GATT_UUID_CHAR_CLIENT_CONFIG;
			descr_uuid.len = ESP_UUID_LEN_16;
	        esp_ble_gatts_add_char_descr(param->add_char.service_handle, &descr_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, NULL, NULL);
		}
		else if(param->add_char.attr_handle == param->add_char.service_handle+5){
			;
		}
		else{
			;
		}
        break;
    }
    case ESP_GATTS_ADD_CHAR_DESCR_EVT:		
        ESP_LOGI(GATTS_TAG, "ADD_DESCR_EVT, status %d, attr_handle %d, service_handle %d\n",
                 param->add_char.status, param->add_char.attr_handle, param->add_char.service_handle);

		/* add a new char to service A */
		if(param->add_char.attr_handle == param->add_char.service_handle+3){
			esp_bt_uuid_t char_uuid;
			char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A2;
			char_uuid.len = ESP_UUID_LEN_16;
			esp_ble_gatts_add_char(param->add_char.service_handle, &char_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, 
									ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY, 
									&gatts_demo_char1_val, NULL);
		}
		
        break;
`
But this way is trouble and inefficiency, as the @TianHao-Espressif advice in gatt_server_service_table add characteristics in a table is more convenient and intuitive.

hmj844263150, 2017-04-24:
I have learn BLE at esp32 nowadays, and got same doubt with you @divinebird, I want to add a new characteristic in a service with gatt_server.
It seems have to create characteristics one by one, and you can get some information with the function:esp_ble_gatts_get_attr_value() by handle you can get in event param. and i have try to coding it:
`case ESP_GATTS_CREATE_EVT:
ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT, status %d, service_handle %d\n", param->create.status, param->create.service_handle);
gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
gl_profile_tab[PROFILE_A_APP_ID].char_uuid.len = ESP_UUID_LEN_16;
gl_profile_tab[PROFILE_A_APP_ID].char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A;

    esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);

    esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid,
                           ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                           ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY, 
                           &gatts_demo_char1_val, NULL);

	
    break;
case ESP_GATTS_ADD_INCL_SRVC_EVT:
    break;
case ESP_GATTS_ADD_CHAR_EVT: {
	uint16_t length = 0;
    const uint8_t *prf_char;
	
	ESP_LOGI(GATTS_TAG, "ADD_CHAR_EVT, status %d,  attr_handle %d, service_handle %d\n",
	                param->add_char.status, param->add_char.attr_handle, param->add_char.service_handle);
    esp_ble_gatts_get_attr_value(param->add_char.attr_handle,  &length, &prf_char);
    ESP_LOGI(GATTS_TAG, "the gatts demo char length = %x\n", length);
    for(int i = 0; i < length; i++){
        ESP_LOGI(GATTS_TAG, "prf_char[%x] =%x\n",i,prf_char[i]);
    }
	
	if(param->add_char.attr_handle == param->add_char.service_handle+2){
		esp_bt_uuid_t descr_uuid;
		descr_uuid.uuid.uuid16 = ESP_GATT_UUID_CHAR_CLIENT_CONFIG;
		descr_uuid.len = ESP_UUID_LEN_16;
        esp_ble_gatts_add_char_descr(param->add_char.service_handle, &descr_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, NULL, NULL);
	}
	else if(param->add_char.attr_handle == param->add_char.service_handle+5){
		;
	}
	else{
		;
	}
    break;
}
case ESP_GATTS_ADD_CHAR_DESCR_EVT:		
    ESP_LOGI(GATTS_TAG, "ADD_DESCR_EVT, status %d, attr_handle %d, service_handle %d\n",
             param->add_char.status, param->add_char.attr_handle, param->add_char.service_handle);

	/* add a new char to service A */
	if(param->add_char.attr_handle == param->add_char.service_handle+3){
		esp_bt_uuid_t char_uuid;
		char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A2;
		char_uuid.len = ESP_UUID_LEN_16;
		esp_ble_gatts_add_char(param->add_char.service_handle, &char_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, 
								ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY, 
								&gatts_demo_char1_val, NULL);
	}
	
    break;`
But this way is trouble and inefficiency, as the @TianHao-Espressif advice in gatt_server_service_table add characteristics in a table is more convenient and intuitive.

hmj844263150, 2017-04-24:
I have learn BLE at esp32 nowadays, and got same doubt with you @divinebird, I want to add a new characteristic in a service with gatt_server.
It seems have to create characteristics one by one, and you can get some information with the function:esp_ble_gatts_get_attr_value() by handle you can get in event param. and i have try to coding it:
`case ESP_GATTS_CREATE_EVT:
        ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT, status %d,  service_handle %d\n", param->create.status, param->create.service_handle);
        gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.len = ESP_UUID_LEN_16;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A;

        esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);

        esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid,
                               ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                               ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY, 
                               &gatts_demo_char1_val, NULL);
        break;
    case ESP_GATTS_ADD_INCL_SRVC_EVT:
        break;
    case ESP_GATTS_ADD_CHAR_EVT: {
		uint16_t length = 0;
        const uint8_t *prf_char;
		
		ESP_LOGI(GATTS_TAG, "ADD_CHAR_EVT, status %d,  attr_handle %d, service_handle %d\n",
		                param->add_char.status, param->add_char.attr_handle, param->add_char.service_handle);
        esp_ble_gatts_get_attr_value(param->add_char.attr_handle,  &length, &prf_char);
        ESP_LOGI(GATTS_TAG, "the gatts demo char length = %x\n", length);
        for(int i = 0; i < length; i++){
            ESP_LOGI(GATTS_TAG, "prf_char[%x] =%x\n",i,prf_char[i]);
        }
		
		if(param->add_char.attr_handle == param->add_char.service_handle+2){
			esp_bt_uuid_t descr_uuid;
			descr_uuid.uuid.uuid16 = ESP_GATT_UUID_CHAR_CLIENT_CONFIG;
			descr_uuid.len = ESP_UUID_LEN_16;
	        esp_ble_gatts_add_char_descr(param->add_char.service_handle, &descr_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, NULL, NULL);
		}
		else if(param->add_char.attr_handle == param->add_char.service_handle+5){
			;
		}
		else{
			;
		}
        break;
    }
    case ESP_GATTS_ADD_CHAR_DESCR_EVT:		
        ESP_LOGI(GATTS_TAG, "ADD_DESCR_EVT, status %d, attr_handle %d, service_handle %d\n",
                 param->add_char.status, param->add_char.attr_handle, param->add_char.service_handle);
		/* add a new char to service A */
		if(param->add_char.attr_handle == param->add_char.service_handle+3){
			esp_bt_uuid_t char_uuid;
			char_uuid.uuid.uuid16 = GATTS_CHAR_UUID_TEST_A2;
			char_uuid.len = ESP_UUID_LEN_16;
			esp_ble_gatts_add_char(param->add_char.service_handle, &char_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, 
									ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY, 
									&gatts_demo_char1_val, NULL);
		}
        break;
`
But this way is inefficiency, as the @TianHao-Espressif advice in gatt_server_service_table add characteristics in a table is more convenient and intuitive.

FayeY, 2017-05-02:
Hi divinebird, is this problem solved? Can we close this issue?

closed 2017-06-19 by FayeY