espressif/esp-idf #285: How to configure GPIO 32 and 33 as input or output in ESP32

amallal, 2017-01-23:
Hi
How i can configure GPIO 32 and 33 as input or output in ESP32, I tried for all other gpio, those are working, Can anyone give some suggestion.

Thanks

negativekelvin, 2017-01-23:
Use gpio_config or call rtc_gpio_deinit first

Possibly similar to https://github.com/espressif/esp-idf/issues/245

nkolban, 2017-01-23:
What does your code fragment look like today for setting GPIOs?  It is likely that you are running into issues because a uint32_t is 32 bits in length (eg. pins 0-31) but when you are setting GPIO 32 or 33, that won't fit in a bitmask that is only 32 bits long.  gpio_config() is the ideal way to go or else use gpio_pad_select_gpio().

amallal, 2017-01-24:
Hi
Thanks for the replies, I checked with 'gpio_pad_select_gpio() followed by ' 'rtc_gpio_deinit'  operation.Now both GPIO 32&33 also i am able to use as gpios.

nkolban, 2017-01-25:
Howdy Amallal, Great news.  Remember to close the issue if you no longer need a response and you consider the issue closed ... or else update the issue with any continuance on the same subject.

closed 2017-04-05 by igrr

ConstantinoSchillebeeckx, 2017-08-15:
To be a bit more explicit with regards to what @nkolban said (since I just struggled through this), you can setup your GPIO pins > 32 with something like this:
```C
// We use unsigned long here since 32 is a long
#define GPIO_BIT_MASK  ((1ULL<<GPIO_NUM_32) | (1ULL<<GPIO_NUM_33)) 

app_main() {
	gpio_config_t io_conf;
	io_conf.intr_type = GPIO_INTR_DISABLE;
	io_conf.mode = GPIO_MODE_OUTPUT;
	io_conf.pin_bit_mask = GPIO_BIT_MASK;
	io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
	io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
	gpio_config(&io_conf);
}
```

🌟

MartyMacGyver, 2019-07-09:
FWIW, if you came here like I did trying to see why GPIO32 and GPIO33 won't work right specifically on the **ESP-WROVER-KIT (with LCD)** despite this, it's because in that particular implementation [those pins are being used for an external crystal](https://dl.espressif.com/dl/schematics/ESP-WROVER-KIT_V4_1.pdf).

ORParga, 2020-05-28:
Before an ESP32 pin can be used as input or output. In ESP32 framework (not Arduino framework) this must be configured by calling `gpio_config ()` This function accepts as the only parameter a `gpio_config_t `structure with five members that must be completely filled. 

To set a pin as output, you must disable the modes "Interruption" "pull Up" and "pull Down", which would be rather input modes. You must configure the mode as "output mode" and you must specify which pins these changes affect using `pin_bit_mask`

`gpio_config_t io_conf;
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;//disable interrupt
    io_conf.mode = GPIO_MODE_OUTPUT;//set as output mode
    io_conf.pin_bit_mask = (1ULL<<18);//bit mask of the pins that you want to set,e.g.GPIO18
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;//disable pull-down mode
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;//disable pull-up mode

    esp_err_t error=gpio_config(&io_conf);//configure GPIO with the given settings

    if(error!=ESP_OK){
        printf("error configuring outputs \n");
    }`

To set a pin as input, You must put the mode ( `io_conf.mode` )as input and select if you want the low level to be interpreted as active or inactive using `pull_down_en`

    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;//disable interrupt
    io_conf.mode = GPIO_MODE_INPUT;//set as inputmode
    io_conf.pin_bit_mask = (1ULL<<15);//bit mask of the pins that you want to set,e.g.GPIO15
    io_conf.pull_down_en = GPIO_PULLDOWN_ENABLE;//enable pull-down mode
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;//disable pull-up mode
    esp_err_t error=gpio_config(&io_conf);//configure GPIO with the given settings

    if(error!=ESP_OK){
        printf("error configuring inputs\n");
    }

to turn on the output 18: `gpio_set_level(18, cnt );` 
to detect the input 15:`int level=gpio_get_level(15);`

take a look at the [espressif official website about GPIO](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html)

MartyMacGyver, 2020-05-29:
Right, but if your device has a crystal on those lines like the wrover kit does, it won't work no matter what you do.

whitevast, 2020-07-23:
> Right, but if your device has a crystal on those lines like the wrover kit does, it won't work no matter what you do.
What development boards can you use these pins on? WROOM?

MartyMacGyver, 2020-07-23:
> What development boards can you use these pins on? WROOM?

I'm not sure what WROOM-based board you are referring to... per [the DEVKIT-C schematics](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf), it doesn't appear to use an external crystal.

OTOH looking at [the WROVER-KIT schematic](https://dl.espressif.com/dl/schematics/ESP-WROVER-KIT_V4_1.pdf) near the top right, we see the crystal (and a cryptic note about certain resistors R36-R39 that doesn't make much sense to me). Short of desoldering things that one's using an external crystal on the usual GPIO 33/34 pins.

So, check the schematics for the board of interest to see if it uses an external crystal or not. It seems like they usually don't - I have no idea why they bother to on the WROVER-KIT except perhaps for possibly tighter timing.

whitevast, 2020-07-24:
> > What development boards can you use these pins on? WROOM?
> 
> So, check the schematics for the board of interest to see if it uses an external crystal or not. It seems like they usually don't - I have no idea why they bother to on the WROVER-KIT except perhaps for possibly tighter timing.

It turns out that on the ESP32 DEVKIT V1 WROOM board you will not be able to control pins 32 and 33?

MartyMacGyver, 2020-07-24:
Is this the exact board you're talking about? https://github.com/espressif/arduino-esp32/issues/544

I don't see any crystal on that particular schematic so I'd think GPIO32 and 33 would work fine.

If this is not the board you're talking about, share the schematic and let's take a look.

whitevast, 2020-07-24:
> Is this the exact board you're talking about? [espressif/arduino-esp32#544](https://github.com/espressif/arduino-esp32/issues/544)
> 
> I don't see any crystal on that particular schematic so I'd think GPIO32 and 33 would work fine.
> 
> If this is not the board you're talking about, share the schematic and let's take a look.

Yes, the board looks exactly the same from above. Only the bottom is black :) But digitalWrite (32, HIGH);     digitalWrite (33, HIGH); does not work.
Even if you do pinMode this way:
```
void aPinMode(int pinNum, int pinDir) {
  // Enable GPIO32 or 33 as output.
  if (pinNum == 32 || pinNum == 33) {
    uint64_t gpioBitMask = (pinNum == 32) ? 1ULL << GPIO_NUM_32 : 1ULL << GPIO_NUM_33;
    gpio_mode_t gpioMode = (pinDir == OUTPUT) ? GPIO_MODE_OUTPUT : GPIO_MODE_INPUT;
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.mode = gpioMode;
    io_conf.pin_bit_mask = gpioBitMask;
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&io_conf);
  } else pinMode(pinNum, pinDir);
}
```

waheedi, 2022-10-23:
I was using BIT(GPIO) as a generic format, I exchanged it with 1ULL << GPIO and then alles ok :) @nkolban Thank you for pointing out the 31 Bit mask limit, I just fixed a bug :+1: