espressif/esp-idf #286: xQueueReceive causes LoadProhibited

jakkubu, 2017-01-23:
The code comes from example on queues from [FreeRTOS documentation: example 10](http://www.freertos.org/Documentation/161204_Mastering_the_FreeRTOS_Real_Time_Kernel-A_Hands-On_Tutorial_Guide.pdf).

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_system.h"
#include "nvs_flash.h"

#include "esp_log.h"
static const char *TAG = "test queues";

QueueHandle_t my_queue;

static void vSenderTask( void *pvParameters )
{
    int32_t lValueToSend; BaseType_t xStatus;
    lValueToSend = (int32_t) pvParameters;
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(500));
        xStatus = xQueueSendToBack(my_queue, &lValueToSend, 0);
        if( xStatus != pdPASS )
        {
            ESP_LOGI(TAG, "Could not send to the queue.");
        }
    }
}

static void vReceiverTask( void *pvParameters )
{
    int32_t lReceivedValue;
    BaseType_t xStatus;
    const TickType_t xTicksToWait = pdMS_TO_TICKS(1100);
    while (1) {
        if( uxQueueMessagesWaiting( my_queue ) != 0 ) {
            ESP_LOGI(TAG, "Queue should have been empty!");
        }

        xStatus = xQueueReceive ( my_queue, &lReceivedValue, xTicksToWait );
        if( xStatus == pdPASS ) {
            ESP_LOGI(TAG, "Received = %d", lReceivedValue);
        } else {
            ESP_LOGI(TAG,  "Could not receive from the queue.");
        }
    }
}

void app_main( void )
{
    nvs_flash_init();
    my_queue = xQueueCreate( 5, sizeof( int32_t ) );
    if( my_queue != NULL ) {
        xTaskCreate( vSenderTask, "Sender1", 1000, ( void * ) 100, 1, NULL );
        xTaskCreate( vSenderTask, "Sender2", 1000, ( void * ) 200, 1, NULL );
        xTaskCreate( vReceiverTask, "Receiver", 1000, NULL, 2, NULL );
    } else {
        ESP_LOGI(TAG, "queue not created");
    }
}
```

And it's causing following error:

```
I (6091) test queues: Received = 200
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40083281  PS      : 0x00060033  A0      : 0x400846af  A1      : 0x3ffb8d00  
A2      : 0x00050023  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000005  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000001  A9      : 0x00000001  
A10     : 0x3ffb88f0  A11     : 0x00000000  A12     : 0x3ffb37c4  A13     : 0x40082134  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x40083281:0x3ffb8d00 0x400846af:0x3ffb8d20 0x40084740:0x3ffb8d40 0x400ef9be:0x3ffb8d80
```

igrr, 2017-01-23:
This example allocates 1000 bytes for each task's stack. By default, ESP-IDF uses fully-featured stdio functions (those include printf), which may use considerable amount of stack space. You may either increase stack space from 1000 bytes to, say, 2000 bytes, or enable "Enable 'nano' formatting options for printf/scanf family" option in menuconfig (under Component config > ESP32-specific config).

forthlightning, 2017-01-23:
Is there somewhere to find more information about what this command "Enable 'nano' formatting options for printf/scanf family" actually does? Does it allow for making the stack space smaller? Is the operation of the 'nano' -fied functions the same?

igrr, 2017-01-24:
Yes, menuconfig has the 'help' button which shows the following for this option:

        ESP32 ROM contains parts of newlib C library, including printf/scanf family
        of functions. These functions have been compiled with so-called "nano"
        formatting option. This option doesn't support 64-bit integer formats and C99
        features, such as positional arguments.

        For more details about "nano" formatting option, please see newlib readme file,
        search for '--enable-newlib-nano-formatted-io':
        https://sourceware.org/newlib/README

        If this option is enabled, build system will use functions available in
        ROM, reducing the application binary size. Functions available in ROM run
        faster than functions which run from flash. Functions available in ROM can
        also run when flash instruction cache is disabled.

        If you need 64-bit integer formatting support or C99 features, keep this
        option disabled.

igrr, 2017-04-05:
Closing this, if there are further issues please feel free to reopen.

closed 2017-04-05 by igrr