espressif/esp-idf #287: spi_master driver cannot support transation with Tx length==0

xiangjunhuang, 2017-01-23:
Hi,

I'm having a transation wrap around spi_master as below:
`uint8_t my_transaction(uint8_t sendBytes, uint8_t *pDin, uint8_t recvBytes, uint8_t *pDout){
    esp_err_t ret;
    spi_transaction_t t;
    memset(&t, 0, sizeof(t));

    t.length=sendBytes*8;
    t.rxlength = recvBytes*8;
    t.tx_buffer = pDin;
    t.rx_buffer = pDout;
    ret=spi_device_transmit(handle, &t);
    TEST_ASSERT(ret==ESP_OK);
  return 0;
}`

When sendBytes>0, this function works OK as epxected, but when sendBytes==0, it was not doing anything by calling this function. I believe this is not a right behavior, can someone help take a deep look inside? thanks,

xiangjunhuang, 2017-01-24:
Found the problem and having a fix below looks like a typo to me:

`diff --git a/components/driver/spi_master.c b/components/driver/spi_master.c
index 6162edb..8f6779f 100644
--- a/components/driver/spi_master.c
+++ b/components/driver/spi_master.c
@@ -655,7 +655,7 @@ static void IRAM_ATTR spi_intr(void *arg)
             host->hw->addr=trans->address & 0xffffffff;
         }
         host->hw->user.usr_mosi=(trans->tx_buffer==NULL)?0:1;
-        host->hw->user.usr_miso=(trans->tx_buffer==NULL)?0:1;
+        host->hw->user.usr_miso=(trans->rx_buffer==NULL)?0:1;
 
         //Call pre-transmission callback, if any
         if (dev->cfg.pre_cb) dev->cfg.pre_cb(trans);
`

negativekelvin, 2017-01-24:
what about 
https://github.com/espressif/esp-idf/blob/master/components/driver/spi_master.c#L627
```
if (trans->rxlength < 8*32) {
                //No need for DMA.
                for (int x=0; x < trans->rxlength; x+=32) {
```

Should be length instead of rx_length?

negativekelvin, 2017-01-31:
@Spritetm had a chance to check this?

Spritetm, 2017-02-21:
Should be fixed when the MR for #336 is merged. Closing this issue for now, please discuss in #336 if you have more issues.

closed 2017-02-21 by Spritetm