espressif/esp-idf #289: Discrepancy on macro definition values ARG_MAX and PATH_MAX

nkolban, 2017-01-23:
While compiling some 3rd party code, I found warnings about re-definitions of a couple of symbols, namely `ARG_MAX` and `PATH_MAX`.  Searching ESP-IDF, I found both of these defined in `components/newlib/include/limits.h` and `components/newlib/include/sys/syslimits.h`.

In `syslimits.h` there are no redefinition guards.  The values of the pairs of variables are not consistent.  In one ARG_MAX = 4096 while in another ARG_MAX = 65536.  And also PATH_MAX = 4096 while in the other PATH_MAX = 1024.

lakidd, 2017-03-19:
Did you ever find a solution to this @nkolban ? I've got the same thing with the pcbreflux examples

nkolban, 2017-03-20:
Thankfully, the warnings are literally that ... warnings ... and other than not getting a pretty compile, all seems to function.  Hopefully the folks at Espressif will continue to track this issue and in good time will offer us an "out of the box" fix.  If you need an immediate workaround, I'd suggest picking one of the header files (probably syslimits.h) and comment out the duplicate symbols.  This will produce a clean compile.  The down-side is that when you bring in a new version of the ESP-IDF, the changes will be undone.

lakidd, 2017-03-21:
Thanks, I added redefinition guards to syslimits.h and the warnings have gone. I thought it was causing some other issues but wasn't the case

closed 2017-05-05 by igrr