espressif/esp-idf #290: TLS/SSL failure with Blufi

seopyoon, 2017-01-24:
Hi, I am trying to integrate Blufi AND tls/ssl connection with AWS IoT at the same time.

When run separately, they work very well with no problem at all.  But, when merged together, I get the following error:
```
DEBUG:   iot_tls_connect L#181   . Setting up the SSL/TLS structure...
ERROR: iot_tls_connect L#206  failed
  ! mbedtls_ssl_setup returned -0x7f00
```
When looked closelyl, the above error code is associated with `MBEDTLS_ERR_SSL_ALLOC_FAILED` (Memory allocation failed) [LINK](https://tls.mbed.org/api/ssl_8h.html#af79cb539a0ee6ac20cf9c574f4c3b343).

My speculation is that Blufi uses MOST of the memory available to the chip, and setting up SSL/TLS requires a lot of variables and large stack size, so it is running out of memory.

Is there a way to find out, using esptool perhaps, to see how much of the internal ram/rom my application is using after compilation, and make sure that such out-of-memory is actually occurring?  Or, since it happens in runtime, there is no way to actually confirm that?  And, finally, could there be any solution to solving this issue?  Or is TLS/SSL just impossible with Blufi?  

Thanks a lot in advance.

seopyoon, 2017-01-25:
I figured, perhaps it is the AWS IoT port on esp that may be causing the problem.  
So, I tried it again, with the HTTPS request example from protocols folder.

And, here is the [link](https://gist.github.com/seopyoon/84db519278b91b17a80f80a7e0de5dd4) to code.  

Very few modifications:  
- I copy-pasted the `server_root_cert.pem` file to the main folder, 
- added `COMPONENT_EMBED_TXTFILES := server_root_cert.pem` to the component.mk file,
- brought out `xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);` to the beginning of the task.  (It was because the I got StoreProhibited error, when setting up ssl/tls stuff without wifi having started.

I get error while `Performing the SSL/TLS handshake...`
The weird part is that the error code keeps changing.  I have got `-0x2880`, `-0x3f80`, `-0x3b10`.

Most of them seem to be associated with memory allocation.  

Any solution to fixing this?  Thanks.

projectgus, 2017-01-26:
It does sound like you're running out of RAM. One of the simplest things you can try is to go to `make menuconfig` -> Components -> mbedTLS and turn down the "TLS Maximum Message Length". There will be a limit under which it won't work, but 8192 bytes should be OK and saves you 16KB of heap memory (there's an RX and a TX buffer both the same size)

seopyoon, 2017-01-26:
@projectgus thank you very much for your suggestion.  That will definitely reduce the memory by a huge amount at once.  I shall try that right away.  

Plus, are there other suggestions to how I could be reducing RAM usage?  Is it the Blufi that is using up a lot of RAM, or was that just a minor addition to what I already have?

closed 2017-02-01 by seopyoon

malaimoo, 2017-12-15:
@projectgus 
Hello, thank for your suggestion above,  I tried to change the "TLS Maximum Message Length" to 8192 but it comes to an error " handshake failed: -9984(0x2700)" . Do you have more suggestion? thank a lot.

malaimoo, 2017-12-16:
@projectgus
I stop bluetooth and release the ram, then it run ok. I think the problem is always the ram lackness ,but the mbedTLS log message various ,and lead me to wrong position.

wegunterjr, 2018-07-09:
@malaimoo your suggestion to release the RAM, does it look like this message:    esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);

malaimoo, 2018-07-10:
@wegunterjr Yes