espressif/esp-idf #292: Ubuntu 64-bit can't make flash

donny681, 2017-01-24:
I setup Ubuntu 64-bit and configurate the toolchain of esp32,it  can complie the code but when I input "make flash" ,it shows that it fail to download to the flash.When I use esptool.py to do it and it can works.I'm sure the port which I choose is right.I don't konw  what's something wrong.

> Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
usage: esptool [-h] [--chip {auto,esp8266,esp31,esp32}] [--port PORT]
               [--baud BAUD] [--no-stub]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: 'default_reset' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')
/home/sky/esp/esp-idf/components/esptool_py/Makefile.projbuild:52: recipe for target 'flash' failed
make: *** [flash] Error 2

negativekelvin, 2017-01-24:
esptool maybe not up to date.  Try git submodule update?

projectgus, 2017-02-27:
@donny681, did you make any progress with this? It does seem like the submodule may have been out of date (the `default_reset` option was added to esp-idf's esptool submodule fairly recently.)

igrr, 2017-04-05:
Closing due to no feedback. Please reopen if the issue persists.

closed 2017-04-05 by igrr