espressif/esp-idf #294: Wifi Example Configuration issue.

rajeshchavan88, 2017-01-24:
Hello All,
I am a beginner on esp-idf. I have started working on esp32 just last week. I am trying to run the wifi example code that we have in ..esp-idf\examples\wifi\wpa2_enterprise. I compiled the code and downloaded partitions_singleapp.bin and wpa2-enterprise.bin files @0x10000 and @0x8000 respectively. However i was not able to get the wifi example code up and working on my esp32 board.
Do i need to do some additional things. I have also gone through readme.md file for this example code, however i did not find it much helpful. I am very naive in this wifi world and it seems i am missing some steps. Please help me in making this example code up and working on ESP32 board. Thanks in advance. I have also attached the  serial console output for reference.

Console output:-

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:2876
load:0x40078000,len:4200
load:0x40098000,len:740
entry 0x4009816c
I (27) boot: Espressif ESP32 2nd stage bootloader v. V0.1
I (36) boot: compile time 15:00:13
D (42) boot: mmu set paddr=00000000 count=1
D (49) boot: magic e9
D (52) boot: blocks 04
D (56) boot: spi_mode 02
D (60) boot: spi_speed 00
D (64) boot: spi_size 00
I (68) boot: SPI Speed      : 40MHz
I (75) boot: SPI Mode       : DIO
I (81) boot: SPI Flash Size : 1MB
I (87) boot: Partition Table:
I (92) boot: ## Label            Usage          Type ST Offset   Length
D (104) boot: load partition table entry from 4000(3f404000)
D (112) boot: type=0 subtype=0
I (117) boot:  0 factory          factory app      00 00 00010000 00100000
D (129) boot: load partition table entry from 4020(3f404020)
D (138) boot: type=1 subtype=1
I (142) boot:  1 rfdata           RF data          01 01 00110000 00040000
D (154) boot: load partition table entry from 4040(3f404040)
D (163) boot: type=1 subtype=2
I (167) boot:  2 wifidata         WiFi data        01 02 00150000 00040000
D (179) boot: load partition table entry from 4060(3f404060)
D (188) boot: type=ff subtype=ff
I (193) boot: End of partition table
I (199) boot: Loading app partition at offset 00010000
D (208) boot: mmu set paddr=00010000 count=16
D (215) boot: bin_header: 233 8 2 1 40080bbc
I (222) boot: section 0: paddr=0x00000020 vaddr=0x00000000 size=0x0ffe8 ( 65512)                                                                                                                                                            
D (234) boot: found drom section, map from 00010010 to 3f400010
I (243) boot: section 1: paddr=0x00010010 vaddr=0x3f400010 size=0x0d914 ( 55572)                                                                                                                                                             map
I (256) boot: section 2: paddr=0x0001d92c vaddr=0x3ffb0000 size=0x0246c (  9324)                                                                                                                                                             load
I (273) boot: section 3: paddr=0x0001fda0 vaddr=0x40080000 size=0x00400 (  1024)                                                                                                                                                             load
I (283) boot: section 4: paddr=0x000201a8 vaddr=0x40080400 size=0x1c55c (116060)                                                                                                                                                             load
Fatal exception (28): LoadProhibited
epc1=0x40098167, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x0                                                                                                                                                            0000000

negativekelvin, 2017-01-24:
Flash a new bootloader @1000

rajeshchavan88, 2017-01-25:
Hi,
I have flashed the new boot-loader @0x1000, still I am getting the same error.

michal037, 2017-01-25:
You have got botloader.bin in bootloader directory, try flash with it.

igrr, 2017-01-25:
> partitions_singleapp.bin and wpa2-enterprise.bin files @0x10000 and @0x8000 respectively. 

Shouldn't that be the other way around?

Have you tried flashing using `make flash` command?

rajeshchavan88, 2017-01-25:
Thanks a lot folks. It helped me. I was downloading through flashing tool. I have downloaded the software through make flash command and it works now.

closed 2017-01-25 by rajeshchavan88