espressif/esp-idf #297: ulp I_RD_REG compile error

xiangjunhuang, 2017-01-25:
Clearly it's a typo:

diff --git a/components/ulp/include/esp32/ulp.h b/components/ulp/include/esp32/ulp.h
index f4e37e9..179d1ee 100644
--- a/components/ulp/include/esp32/ulp.h
+++ b/components/ulp/include/esp32/ulp.h
@@ -320,7 +320,7 @@ static inline uint32_t SOC_REG_TO_ULP_PERIPH_SEL(uint32_t reg) {
  * R0 = reg[high_bit : low_bit]
  * This instruction can access RTC_CNTL_, RTC_IO_, and SENS_ peripheral registers.
  */
-#define I_RD_REG(reg, low_bit, high_bit, val) {.wr_reg = {\
+#define I_RD_REG(reg, low_bit, high_bit, val) {.rd_reg = {\
     .addr = reg & 0xff, \
     .periph_sel = SOC_REG_TO_ULP_PERIPH_SEL(reg), \
     .unused = 0, \

xiangjunhuang, 2017-01-25:
I tried to this I_RD_REG in my project, but seems the result I got is incorrect. Can someone please confirm if this is really implemented in esp-idf 2.0 release?

igrr, 2017-01-25:
Thanks for reporting this. We'll fix the issue before v2.0.

xiangjunhuang, 2017-01-26:
Thanks @igrr for your quick response! did you get chance to test after .wr_reg -> .rd_reg? Can it really readback right value from a register? I did this in on my board, but not working as expected somehow.

igrr, 2017-01-26:
There is a bug in the definitions of `I_RD_REG`/`I_WR_REG`, we'll have a fix shortly.

xiangjunhuang, 2017-02-03:
@igrr, even AFTER I fixed the I_RD_REG definition locally, I still cannot read correct value from this instruction. Can you please confirm if that works for you?

closed 2017-03-08 by igrr

igrr, 2017-03-09:
The bugs related to `I_RD_REG` have been fixed in master now. Please give it a try.